carla-simulator/carla #307: Network speed cap between Carla Client and Server

jcyriac2, 2018-03-24:
I have a Carla Imitation Learning Client(given in the carla repo) running on a Linux machine and the Carla Server.exe (Release 0.7.1) running on a Windows 10 machine. The Carla server is run with the fixed-time step set to fps=30.
```.\CarlaUE4.exe /Game/Maps/Town01 -carla-server -benchmark -fps=30 -windowed -ResX=800 -ResY=600 -carla-settings=.\CarlaSettings.ini```
The two machines are connected directly by a 1Gbit/s connection.  The machines are fast with good GPUs.
We are experiencing a network speed cap of about 80MBit/s which is causing the simulation on the server to lag. Is this a bug in the code or is there a way to increase the connection speed between the server and the client so that the simulation runs faster?

[image]

YashBansod, 2018-03-24:
@jcyriac2 Are you sure you are using a CAT6 cable (atleast)? Can you confirm if the resource monitor recognises the connection as a gigabit connection?

nsubiron, 2018-03-24:
Hi @jcyriac2,

There are ways to tune the network to get higher speeds, see for instance [Network / TCP / UDP Tuning](https://wwwx.cs.unc.edu/~sparkst/howto/network_tuning.php).

But could it be that the simulator is slow producing data? How many cameras you added? For instance, one image of 800x600 is about 15MB, at 5 FPS that makes ~75MB/s.

jcyriac2, 2018-03-24:
Thank you for the quick replies.
@YashBansod Yes I have tested the raw network speed between the 2 machines and it is around 935Mbit/s.
@nsubiron We are using the code from the imitation learning library at carla-simulator/imitation-learning. So that has one 800x600 rgb camera. Will just that one camera cause the simulator to run slowly? Also I have noticed that when I run the carla server at a lower fixed timestep (lowered from fps=30 to fps=15), the simulation seems to run faster. Does this fps value directly influence how many camera sensor images are sent to the client?

Thank you for your help

nsubiron, 2018-03-26:
@jcyriac2 Yes, adding cameras slows down the simulation significantly. We have improved the performance of cameras for the 0.8.0 release (the release is not yet published but the master branch here in GitHub already has the changes).

Indeed, if you run with fixed time-step with lower FPS the simulation runs faster as it has to run less updates per second. We are writing the documentation for this, you can see the draft [here](http://carla.readthedocs.io/en/documentation/configuring_the_simulation/).

> Does this fps value directly influence how many camera sensor images are sent to the client per second?

The number of images per second that the simulator generates is (almost) constant, but the elapsed time in between images is higher.

jcyriac2, 2018-03-28:
Got it! Thanks alot for the help!

closed 2018-03-28 by jcyriac2

cjerome0321, 2019-08-13:
> I have a Carla Imitation Learning Client(given in the carla repo) running on a Linux machine and the Carla Server.exe (Release 0.7.1) running on a Windows 10 machine. The Carla server is run with [...]

Hi @jcyriac2,

I am trying to build the same set-up as yours on two separated laptops under the same WiFi environment: 
Windows  (CARLA 0.9.5- server render machine) - Python 3.7.0
Linux  (CARLA 0.9.5- PythonAPI client manual_control.py) - Python 3.5
After I ran:
```
Windows:
.\CarlaUE4.exe  -carla-server
Linux:
python3 manual_control.py --host=192.168.1.18
```
The linux client could not connect to the server and giving a runtime error.  I've checked the bandwidth between two machines using iperf, which is around 10 Mbits/s. And I also checked connection from linux to 192.168.1.18:2000 using telnet, which is fine. Could you please kindly tell me how you set up the two machines and get it working? Thanks in advance!!

jcyriac2, 2019-08-14:
Hey @cjerome0321 

I haven't worked with CARLA 0.9.5. However, I believe that that in the older versions, you could pass in a command-line option to the client to specify the port number of the server using -p.
`python3 manual_control.py --host=192.168.1.18 -p 2000`

You also might want to check if Windows Firewall is blocking the connection through that port and add an exception for the CarlaUE4.exe app.

cjerome0321, 2019-08-15:
> Hey @cjerome0321
>
> I haven't worked with CARLA 0.9.5. However, I believe that that in the older versions, you could pass in a command-line option to the client to specify the port number of the [...]

Hi @jcyriac2 ,

It worked. Thanks.

wuxiaohua1011, 2020-07-03:
although this thread is dead, i also wanted to add that for slow network speed, the time to transfer an OpenDrive `.xodr` file might be larger than 2 seconds and thus causing a time out error, which is quite misleading. 

The fix to this is simply increasing the timeout to something like 5 or 10 seconds, depending on your wifi speed.