carla-simulator/carla #309: Sensor_data empty resulting in black screen in manual_control.py

Ajk4, 2018-03-26:
I am using CARLA_0.7.1

I'm starting Carla Simulator on remote machine via

`./CarlaUE4.sh /Game/Maps/Town01 -carla-server -benchmark -fps=15`

Then I call manual_control.py on my machine to connect to the server

`./manual_control.py --host $REMOTE_HOST --port 2000`

* It starts and I can control my vehicle (and see changing vehicle speed) 
* But the pygame screen remains blank black.
* Received sensor_data object is an empty dict.
* On the server side there are no errors.

However when I check logs on the server.

What is suspicious is that CARLA Settings on the server differ from the one provided by Python script. On the server I can config without any sensors, while they were defined in settings string on the client side.

Did any one had similar problem? Cheers

Server Log after client connecting:

```
[2018.03.26-12.12.03:164][ 15]LogCarla: == CARLA Settings ==============================================================
[2018.03.26-12.12.03:164][ 15]LogCarla: Last settings file loaded: <string-provided-by-client>
[2018.03.26-12.12.03:164][ 15]LogCarla: [CARLA/Server]
[2018.03.26-12.12.03:164][ 15]LogCarla: Networking = Enabled
[2018.03.26-12.12.03:164][ 15]LogCarla: World Port = 2000
[2018.03.26-12.12.03:164][ 15]LogCarla: Server Time-out = 10000 ms
[2018.03.26-12.12.03:164][ 15]LogCarla: Synchronous Mode = Disabled
[2018.03.26-12.12.03:164][ 15]LogCarla: Send Non-Player Agents Info = Enabled
[2018.03.26-12.12.03:164][ 15]LogCarla: [CARLA/LevelSettings]
[2018.03.26-12.12.03:164][ 15]LogCarla: Player Vehicle        = Default
[2018.03.26-12.12.03:164][ 15]LogCarla: Number Of Vehicles    = 15
[2018.03.26-12.12.03:164][ 15]LogCarla: Number Of Pedestrians = 30
[2018.03.26-12.12.03:164][ 15]LogCarla: Weather Id = 8
[2018.03.26-12.12.03:164][ 15]LogCarla: Seed Vehicle Spawner = 29956
[2018.03.26-12.12.03:164][ 15]LogCarla: Seed Pedestrian Spawner = 42264
[2018.03.26-12.12.03:164][ 15]LogCarla: Found 15 available weather settings.
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 0 - Default
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 1 - ClearNoon
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 2 - CloudyNoon
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 3 - WetNoon
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 4 - WetCloudyNoon
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 5 - MidRainyNoon
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 6 - HardRainNoon
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 7 - SoftRainNoon
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 8 - ClearSunset
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 9 - CloudySunset
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 10 - WetSunset
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 11 - WetCloudySunset
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 12 - MidRainSunset
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 13 - HardRainSunset
[2018.03.26-12.12.03:164][ 15]LogCarla:   * 14 - SoftRainSunset
[2018.03.26-12.12.03:165][ 15]LogCarla: [CARLA/SceneCapture]
[2018.03.26-12.12.03:165][ 15]LogCarla: Added 0 cameras.
[2018.03.26-12.12.03:165][ 15]LogCarla: Semantic Segmentation = Disabled
[2018.03.26-12.12.03:165][ 15]LogCarla: ================================================================================
```

nsubiron, 2018-03-26:
@Ajk4 Which version of the "manual_control.py" are you using? If you are running CARLA 0.7.1 you should use the client examples provided with the 0.7.1 package.

Ajk4, 2018-03-26:
Oh. It must have been it! I took latest code for python package from CARLA 0.7.1 and it works!

Thanks!

closed 2018-03-26 by Ajk4