carla-simulator/carla #310: Start Positions are different before the first reset.

felipecode, 2018-03-26:
The CARLA simulator periodically resets itself while waiting for 
the python client to connect.
However, If the python client system connects to the 
CARLA simulator before the server resets a first time, it receives
a different set of starting positions than expected.

Tested:
Carla 0.7.1
Carla 0.8

nsubiron, 2018-04-17:
**First episode**
[image]

**Second and onwards**
[image]

nsubiron, 2018-04-17:
We can sort the positions before sending them by object name, this way the positions would always arrive in the same order.

```diff
diff --git a/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaGameModeBase.cpp b/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaGameModeBase.cpp
index 1e0049d..f191b21 100644
--- a/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaGameModeBase.cpp
+++ b/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaGameModeBase.cpp
@@ -287,5 +287,8 @@ APlayerStart *ACarlaGameModeBase::FindUnOccupiedStartPoints(
 #endif // WITH_EDITOR
     }
   }
+  UnOccupiedStartPoints.Sort([](const APlayerStart &Lhs, const APlayerStart &Rhs){
+    return Lhs.GetName() < Rhs.GetName();
+  });
   return FoundPlayerStart;
 }
```

However, this order would be the one of the first reset, i.e., precisely the one that we don't want. I don't see any other property of the actors persistent among runs and episodes.

Changing the order of the positions would mean to change all the code that relies on that, on benchmark and other client code, where this order is hard-coded.

Since the client-server communication is going to change soon, allowing arbitrary positions and other improvements, it is better not to change this and live with this bug for now.

nsubiron, 2018-07-30:
As from version 0.9.0 we no longer have start positions listed, we'll add at some point a system for recommending start locations.

closed 2018-07-30 by nsubiron