carla-simulator/carla #312: Error using CARLA from VNC

GokulNC, 2018-03-27:
When I run `$CARLA_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4` from my VNC, I get the following error:

```
Fatal error: [File:/home/nsubiron/UnrealEngine_4.17/Engine/Source/Runtime/Core/Private/Linux/LinuxApplication.cpp] [Line: 36] 
FLinuxApplication::CreateLinuxApplication() : PlatformInitMultimedia() failed, cannot create application instance.
[2018.03.27-05.32.59:748][  0]LogLinux: Error: appError called: Assertion failed: Assertion failed:  [File:/home/nsubiron/UnrealEngine_4.17/Engine/Source/Runtime/Core/Private/Linux/LinuxApplication.cpp] [Line: 36] 
FLinuxApplication::CreateLinuxApplication() : PlatformInitMultimedia() failed, cannot create application instance.

Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
```

**Details**:
OS: Ubuntu 16.04
CARLA: Release version 0.7.0

Also, if you notice, why does it have @nsubiron 's path? ;)

felipecode, 2018-03-27:
Hey @GokulNC 
Thanks for the issue.

Can you tell me exactly which commands you executed ?
Are you using VGL as well ?
Are using our tutorial ? http://carla.readthedocs.io/en/latest/carla_headless/
It seems your vnc didn't work, the display is not available.

Cheers.

GokulNC, 2018-03-28:
Thanks @felipecode for you quick reply.

I solved the problem.
Here are the steps I took:

Since I was running it on an headless server, I installed Ubuntu Desktop package by:
`sudo apt-get install ubuntu-desktop`
(Since many virtual desktop environments like `Xfce` didn't seem to work fine; I faced the above error)

(Also, the X display manager that worked for me was `lightdm`)

Then thought of using that real desktop directly instead of virtual desktops.
So I was able to do that using the answer given here: [How to access an ubuntu machine via VNC from the login screen?](https://askubuntu.com/questions/35396/how-to-access-an-ubuntu-machine-via-vnc-from-the-login-screen/592538#592538)

So I had to run the script `./start_x11vnc.sh`, by which a x11vnc session starts at port 5900.
Now I could connect directly using any VNC Viewer like Real VNC and access the Desktop GUI.

By default, CARLA runs on GPU 0. To specify which GPU to use, I had to run it using `vglrun` [as described in the docs](http://carla.readthedocs.io/en/latest/carla_headless/).

So, I'm closing this issue :)

I think apart from the docs, this is also another way to run CARLA on a headless server.
Do you think it can be added to the docs (since it's actually very simple steps, and guaranteed to work)?

Also, just out of curiosity, why did my error logs have nsubiron's path? ;)

closed 2018-03-28 by GokulNC

chankim, 2018-11-03:
Hello, I exactly followed the doc : https://carla.readthedocs.io/en/latest/carla_headless/
but when I connect using RealVNC from windows 10 to my machine xx.yy.cc.dd:8, I get just the purple screen. seems like there is no windowmanager running.  I understand the last command "DISPLAY=:8 vglrun -d :7.<gpu_number> $CARLA_PATH/CarlaUE4/Binaries/Linux/CarlaUE4" should be run on the vnc window. Am I correct? (I mean, after this window manager problem is solve)

nuomizai, 2020-06-18:
> Hey @GokulNC
> Thanks for the issue.
> 
> Can you tell me exactly which commands you executed ?
> Are you using VGL as well ?
> Are using our tutorial ? http://carla.readthedocs.io/en/latest/carla_headless/
> It seems your vnc didn't work, the display is not available.
> 
> Cheers.

Hey @felipecode , the doc link http://carla.readthedocs.io/en/latest/carla_headless/ does't work anymore. Does carla still support running from headless?