carla-simulator/carla #314: How to display 3D bounding box in real-time

xchhuang, 2018-03-27:
While running the client_example.py, we can get non-players measurements including their locations and box extents. So here we can have bounding boxes for non-players.

But does anyone know how to display/render these bounding boxes in real-time when running client_example.py ( when the car is running ) just like doing 3D detections ?

Thanks.

marcgpuig, 2018-04-06:
Hi @xchhuang, sorry for the late response.

In fact you have the information of these points and the extrinsic and intrinsic matrix.

First you can get these matrices:

Outside the game loop you can obtain the `camera_to_car_transform` matrix and the intrinsic `K` matrix:

```py
# Compute the camera transform matrix
camera_to_car_transform = camera0.get_unreal_transform()

# (Intrinsic) (3, 3) K Matrix
K = np.identity(3)
K[0, 2] = camera0.width / 2.0
K[1, 2] = camera0.height / 2.0
K[0, 0] = K[1, 1] = camera0.width / (2.0 * tan(FOV * pi / 360.0))
```

Inside the game loop you can obtain the extrinsic (Rt) matrix.

```py
# (Camera) local 3d to world 3d
# Get the transform from the player protobuf transformation
world_transform = Transform(
    measurements.player_measurements.transform
)
# Compute the final transformation matrix
Rt = world_transform * camera_to_car_transform
# Get the (4, 4) Numpy matrix itself
Rt = Rt.matrix
```

Now you can iterate each `non_player_agent` and, for instance, project its 3d position to your 2d image:

```py
for agent in self._measurements.non_player_agents:
    if agent.HasField('vehicle'): # if you want only cars
        pos = agent.vehicle.transform.location
        pos_vector = np.array([[pos.x], [pos.y], [pos.z], [1.0]])

        trnasformed_3d_pos = np.dot(inv(Rt), pos_vector)
        pos2d = np.dot(K, trnasformed_3d_pos[:3])

        # Normalize the point 
        pos2d = np.array([
            pos2d[0] / pos2d[2],
            pos2d[1] / pos2d[2],
            pos2d[2]])

        # Now, pos2d contains the [x, y, d] values of the image in pixels (where d is the depth)
        # You can use the depth know the points that are in front of the camera (positive depth).
        if pos2d[2] > 0:
            x_2d = WINDOW_WIDTH - norm_pos2d[0]
            y_2d = WINDOW_HEIGHT - norm_pos2d[1]
            # Draw something in the image 
```

I modiefied a [client](https://gist.github.com/marcgpuig/8317fdb8ffd2a01f7156417d29d05857) so you can try it by yourself. It uses the [0.8.1](https://github.com/carla-simulator/carla/releases/tag/0.8.1) version of CARLA.

You can now modify the code to transform the bounding boxes instead of the 3d position.

marcgpuig, 2018-04-09:
Hi @wzhAptiv,
We don't have this feature right now.
But I think you could make an approach using the depth of the image and the distance from the camera to each agent.

wzhAptiv, 2018-05-03:
Hi @xchhuang , finally, I found the to_unreal_transform() is incorrect for LiDAR sensor

You should override the function in carla.sensor.Lidar as 

```python
def get_unreal_transform(self): 
    to_unreal_transform = Transform(Rotation(yaw=90), Scale(z=-1))
    return self.get_transform() * to_unreal_transform`
```

@marcgpuig I think this is a bug, I'll post an new issue for it.

closed 2018-05-30 by marcgpuig

lxtttt, 2018-06-11:
@wzhAptiv Hi, do you solve the render problem? Could you give me some hints on reducing the unwanted bounding boxes?

wzhAptiv, 2018-06-11:
Hi @lxtttt , I can't locate the render problem in this issue, could you please explain the problem again? 
For the unwanted bbox reduction, you can follow the suggestion by @marcgpuig , using the depth sensor and the distance of object.

whbcumtb, 2019-03-12:
@marcgpuig 
      I am reading the code `world_transform = Transform(
    measurements.player_measurements.transform
)` in your reply commented on 7 Apr 2018, and have a question that how to get the `world_transform`  in the latest version (version 0.9.4) on windows platform, can you give me some detailed information?  

In the previous version, there is a way to read the data produced by the server according to the function  `measurements, sensor_data = client.read_data()` from the `carla_server_pb2.py` and `tcp.py,` while these two files are deprecated in the latest version. 

And we reuse them in the PythonAPI folder, but there is an error that `tcp.TCPConnectionError:(127.0.0.1:2001) not connected`, our code is as follows:
`_stream_client = tcp.TCPClient(host, world_port + 1, timeout)`

kartik144, 2019-04-26:
Hi @marcgpuig ,
I am trying to get the intrinsic matrix in version `0.9.3` but am getting a few errors when trying your code given in this [code](https://gist.github.com/marcgpuig/8317fdb8ffd2a01f7156417d29d05857), possibly due to version mismatch.

I have two questions:
1. What is the equavalent of `get_unreal_transform` in `0.9.3`? is it  just `get_transform`?
1. How can we access the attributes like `fov` and `image_size_x` of an RGB camera set up? I saw in the API guide that we can do it through `get_attribute` of `carla.ActorBlueprint`, but I do not understand how to get that of an actor that has already been created.

Also, it would be really great if you could tell any other changes in the code that you have given to get the extrinsic and intrinsic matrices that have to be made to use it in version `0.9.3`.
Essentially, I just want to get the intrinsic and extrinsic matrices for an RGB camera for point cloud registration.