carla-simulator/carla #318: Customized CameraRGB size not working

wzhAptiv, 2018-03-28:


juaxix, 2018-03-29:
Hello @wzhAptiv ,can you provide the command and log please?,
thanks

wzhAptiv, 2018-03-29:
Hello @juaxix , here are the command for starting carla-server, the modified manual_control.py (only changed WINDOW_WIDTH = 400, WINDOW_HEIGHT = 300, MINI_WINDOW_WIDTH = 160, MINI_WINDOW_HEIGHT = 90), and the python command with cmd-output. 
I only omitted the absolute path as 'Path', please be aware. 

[run_Carla.bat.txt](https://github.com/carla-simulator/carla/files/1859888/run_Carla.bat.txt)
[manual_control_demo_debug.py.txt](https://github.com/carla-simulator/carla/files/1859889/manual_control_demo_debug.py.txt)
[python_cmd_log.txt](https://github.com/carla-simulator/carla/files/1859890/python_cmd_log.txt)

I tried a few other window_widths & heights, non of them worked except 800x600. 
The error came from 'CameraRGB'. 

Thank you.

juaxix, 2018-03-29:
Hey @wzhAptiv I noticed that your command has a typo with the option _-carla-server_ ,you used -clara-server, please fix that.

Now, I've just tested it with this command for the server:
`CarlaUE4.exe /Game/Maps/Town01 -windowed -ResX=400 -ResY=300 -carla-server -carla-world-port=2000 -fps=10 -benchmark`

and this command for the client:
`python manual_control.py -l -q Low -m Town01`

and this works perfectly as you can see in the screenshots:
Town 1:
[image]

Town 2:
[image]

I'm using the python version 3.6.4

Note: if you are not going to use the main screen you can set this command (use º to bring up the console in the server main window) 

> r.ScreenPercentage 1

and with the G key you can deactivate the HUD, that will give you around 2-3FPS more.
Thanks

wzhAptiv, 2018-03-29:
Hello @juaxix , have you tried to change the window size? or try my modified .py script? 
According to your screenshot, it seems you are just running 800x600, which I also have no problem. 

Also, thanks for pointing out the typo in command.

juaxix, 2018-03-29:
You are right @wzhAptiv ,this is a bug related to Direct3D that is adding more information to the texture of the screen capture to be sent so the array of data is bigger that 4 colors * width * height. 
The work-around right now is go and try to use other API instead of Direct3D.

Run:

> CarlaUE4.exe /Game/Maps/Town01 -windowed -ResX=400 -ResY=300 -carla-server -carla-world-port=2000 -fps=10 -benchmark  **-opengl** 

or 

> CarlaUE4.exe /Game/Maps/Town01 -windowed -ResX=400 -ResY=300 -carla-server -carla-world-port=2000 -fps=10 -benchmark **_-vulkan_** 

because CARLA for Windows is built with support for OpenGL3/4 (SM4,SM5) or Vulkan (SM5). 

You can install Vulkan drivers for Windows using [intel](https://software.intel.com/en-us/blogs/2016/03/14/new-intel-vulkan-beta-1540204404-graphics-driver-for-windows-78110-1540), [nvidia](https://developer.nvidia.com/vulkan-driver) or [amd](https://www.amd.com/en/technologies/vulkan) support pages.

Screenshot using Vulkan:
[image]

We will dig in this to get more info of this with the EPIC team to find a fix for Unreal 4.19 together with other fixes.

wzhAptiv, 2018-03-29:
Thank you and I'll try other API at the moment. 
Hope we can see the fix soon.

juaxix, 2018-04-05:
Update:
The problem is this line

` uint8 *src = reinterpret_cast<uint8*>(RHILockTexture2D(texture, 0, RLM_ReadOnly, stride, false));`

With a resolution of 320x180 the stride is 1280 so `1280*180` = 230400 --> it's the same as `320*180*4`  = 230400

With 400x300 the stride is 1664 -->  `1664*300` = 499200 != `400*300*4` = 480000

With 800x600 the stride is 3200 --> `3200*600` = 1920000 = `800*600*4`

so you just have to find the correct resolution

juaxix, 2018-04-10:
This is fixed in b8b27447d94a3f6877d896ef8422c991d2a4ec9c

wzhAptiv, 2018-04-11:
Thank you for the fix!

commit b8b27447 (2018-04-10): Fix #318 - Direct3D RHI textures resampled to maintain the width*height*bytes_per_pixels being sent to the client when dest_stride!=src_stride
PR #333 (2018-04-13): Optimizations
Some fixes included in this pull request.

  - [x] Walkers uses a closer angle to detect vehicles and the algorithm is changed (will be moved again to async find path)
  - [x] Changes in code formatting
 
#### Description

Most of the changes are related to settings, pedestrian AIs and vehicle spawning

Fixed #246
Fixed #318
Fixed #330 
Fixed #334

  - [x] Minor bugs fixed
    - [x] Vehicles spawner spawns correctly the number specified in the configuration (carla settings file) using a timer to spawn more if there is no space at the start
    - [x] Changes in the screen captures [... cut]
closed 2018-04-13 by nsubiron