carla-simulator/carla #319: Can I run PythonClient with CarlaUE4Editor

maverick-long, 2018-03-28:
Dear all,

After editing the map and making some changes in the carla source code, how can I launch the carla simulation server through UE4Editor and connect it to one of the python clients (for example, ./manual_control.py)? When I hit 'play' in the UE4Editor, the simulation would directly start running under teleoperation mode. 

Thank you very much!

maverick-long, 2018-03-28:
Okay, I figured that I only need to change the 'UseNetworking' entry from false to true.

closed 2018-03-28 by maverick-long

Super-Tree, 2018-05-17:
Hi,
the same question occurred to me,and I don't quite understand what you said above about changing the flag 'UseNetworking'. Do you mean that modifing the file '~/Unreal/CarlaUE4/Config/CarlaSettings.ini' with changing  UseNetworking = false to UseNetworking = true?  and then? Re-run Setup.sh?or Re-run the whole building process? Thanks a lot  if you could give me some help!

gaurish123, 2019-02-19:
Hi,

How will we make sure that CarlaUE4 is listening on port 2000. Also I am using Windows-10. Is the behaviour diff in it?