carla-simulator/carla #324: Adjust the position and rotation of camera when a new episode starts.

wangzhiqing, 2018-04-01:
Suppose there is only one camera, and I want to adjust the position and rotation of the camera after a new episode starts. How can I do this?

YashBansod, 2018-04-02:
[https://github.com/carla-simulator/carla/blob/master/PythonClient/manual_control.py](url)

here you can make changes to this section:

```
    def _on_new_episode(self):
        self._carla_settings.randomize_seeds()
        self._carla_settings.randomize_weather()
        scene = self.client.load_settings(self._carla_settings)
        number_of_player_starts = len(scene.player_start_spots)
        player_start = np.random.randint(number_of_player_starts)
        print('Starting new episode...')
        self.client.start_episode(player_start)
        self._timer = Timer()
        self._is_on_reverse = False
```
The following line is responsible for reading the settings:
`scene = self.client.load_settings(self._carla_settings)`

nsubiron, 2018-04-03:
@wangzhiqing As @YashBansod said, you should call `load_settings` before `start_episode` to load a new configuration. The easiest would be to create a new `CarlaSettings` object and add a camera to it at the desired position.

Minimal example

```python
with make_carla_client(host, port) as client:
    for episode in range(0, number_of_episodes):

        settings = CarlaSettings()
        camera = Camera('MyCamera')
        camera.set_position(0.30, 0.0, 1.0 + episode * 0.20)
        settings.add_sensor(camera)

        client.load_settings(settings)
        client.start_episode(0) # start at first start spot.

        for frame in range(0, frames_per_episode):
            measurements, sensor_data = client.read_data()
            client.send_control(measurements.player_measurements.autopilot_control)
```

closed 2018-04-03 by wangzhiqing