carla-simulator/carla #325: trying carla_ros_bridge

Todspaniard, 2018-04-01:


laurent-george, 2018-04-03:
Hi, 

I have updated the readme (it is maybee more clear now). Could you retry it ? You need to source the catkin workspace before using the roslaunch (and you need to be able to import carla from python2.7, to be able to do that I add a line that use pip to install the carla python client). 

Moreover as this code is not yet in the carla repository, it could be better to add issues related to the rosbridge here: https://github.com/laurent-george/carla (in order to avoid pollutting the carla issues).

Todspaniard, 2018-04-03:
I've tried as you revised,but another error message comming up and can not resolve yet.
the message is
raise RuntimeError('cannot import "carla_server_pb2.py", run '
RuntimeError: cannot import "carla_server_pb2.py", run the protobuf compiler to generate this file

detail is following.

~$ roslaunch ~/carla/carla_ros_bridge/client.launch
started roslaunch server http://XXXXX-ILeDXi-C122:40279/

SUMMARY
========

PARAMETERS
 * /carla/QualityLevel: Low
 * /carla/SynchronousMode: True
 * /carla/host: localhost
 * /carla/port: 2000
 * /carla/sensors/CameraRGB/SensorType: CAMERA
 * /carla/sensors/CameraRGB/carla_settings/FOV: 90
 * /carla/sensors/CameraRGB/carla_settings/ImageSizeX: 800
 * /carla/sensors/CameraRGB/carla_settings/ImageSizeY: 600
 * /carla/sensors/CameraRGB/carla_settings/PositionX: 0.0
 * /carla/sensors/CameraRGB/carla_settings/PositionY: 0
 * /carla/sensors/CameraRGB/carla_settings/PositionZ: 1.0
 * /carla/sensors/CameraRGB/carla_settings/PostProcessing: SceneFinal
 * /carla/sensors/CameraRGB/carla_settings/RotationPitch: 0
 * /carla/sensors/CameraRGB/carla_settings/RotationRoll: 0
 * /carla/sensors/CameraRGB/carla_settings/RotationYaw: 0
 * /carla/sensors/camera_depth/SensorType: CAMERA
 * /carla/sensors/camera_depth/carla_settings/FOV: 90
 * /carla/sensors/camera_depth/carla_settings/ImageSizeX: 800
 * /carla/sensors/camera_depth/carla_settings/ImageSizeY: 600
 * /carla/sensors/camera_depth/carla_settings/PositionX: 0.0
 * /carla/sensors/camera_depth/carla_settings/PositionY: 0
 * /carla/sensors/camera_depth/carla_settings/PositionZ: 1.0
 * /carla/sensors/camera_depth/carla_settings/PostProcessing: Depth
 * /carla/sensors/camera_depth/carla_settings/RotationPitch: 0
 * /carla/sensors/camera_depth/carla_settings/RotationRoll: 0
 * /carla/sensors/camera_depth/carla_settings/RotationYaw: 0
 * /carla/sensors/camera_semantic_segmentation/SensorType: CAMERA
 * /carla/sensors/camera_semantic_segmentation/carla_settings/FOV: 90
 * /carla/sensors/camera_semantic_segmentation/carla_settings/ImageSizeX: 800
 * /carla/sensors/camera_semantic_segmentation/carla_settings/ImageSizeY: 600
 * /carla/sensors/camera_semantic_segmentation/carla_settings/PositionX: 0.0
 * /carla/sensors/camera_semantic_segmentation/carla_settings/PositionY: 0
 * /carla/sensors/camera_semantic_segmentation/carla_settings/PositionZ: 1.3
 * /carla/sensors/camera_semantic_segmentation/carla_settings/PostProcessing: SemanticSegmentation
 * /carla/sensors/camera_semantic_segmentation/carla_settings/RotationPitch: 0
 * /carla/sensors/camera_semantic_segmentation/carla_settings/RotationRoll: 0
 * /carla/sensors/camera_semantic_segmentation/carla_settings/RotationYaw: 0
 * /carla/sensors/lidar_0/SensorType: LIDAR_RAY_CAST
 * /carla/sensors/lidar_0/carla_settings/Channels: 4
 * /carla/sensors/lidar_0/carla_settings/LowerFovLimit: -1.6
 * /carla/sensors/lidar_0/carla_settings/PointsPerSecond: 200000
 * /carla/sensors/lidar_0/carla_settings/PositionX: 0.0
 * /carla/sensors/lidar_0/carla_settings/PositionY: 0
 * /carla/sensors/lidar_0/carla_settings/PositionZ: 1.5
 * /carla/sensors/lidar_0/carla_settings/Range: 120
 * /carla/sensors/lidar_0/carla_settings/RotationFrequency: 25
 * /carla/sensors/lidar_0/carla_settings/RotationPitch: 0
 * /carla/sensors/lidar_0/carla_settings/RotationRoll: 0
 * /carla/sensors/lidar_0/carla_settings/RotationYaw: 0
 * /carla/sensors/lidar_0/carla_settings/UpperFovLimit: 1.6
 * /carla_autopilot: True
 * /enable_rosbag: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

ODES
  /
    carla_ros_bridge (carla_ros_bridge/client.py)

auto-starting new master
process[master]: started with pid [5487]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 72722cf4-373d-11e8-bff1-309c230a0502
process[rosout-1]: started with pid [5500]
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [5517]
Traceback (most recent call last):
  File "/home/xxxxx/catkin_ws/src/carla_ros_bridge/src/client.py", line 22, in <module>
    from carla.client import make_carla_client
  File "/home/xxxxx/carla/PythonClient/carla/client.py", line 14, in <module>
    from . import sensor
  File "/home/xxxxx/carla/PythonClient/carla/sensor.py", line 19, in <module>
    from .transform import Transform, Translation, Rotation, Scale
  File "/home/xxxxx/carla/PythonClient/carla/transform.py", line 20, in <module>
    raise RuntimeError('cannot import "carla_server_pb2.py", run '
RuntimeError: cannot import "carla_server_pb2.py", run the protobuf compiler to generate this file

ps .
I checked dependancy including protobuf and so on  for Python2 and could be ok. but still an issue as above.

laurent-george, 2018-04-03:
I rerun the full readme on a clean linux machin (I have again updated a bit), and everything works fine. 

- Are you using a virtualenv ? (I am not, mixing ros + virtualenv not sure it works correctly).
- Are you able to do something like: from carla import carla_server_pb2 
in a python2 shell ? 
- Are you mixing multiple carla directory ?
- Are you able to run the client.py without ros but using the same python ?

maybee someone from carla team could help.

Todspaniard, 2018-04-03:
Thanks a lot for your quick inquiry.
1.  Virturalenv  :   no i do not use.
2. i can not do that  on python2 shell (from carla import carla_server_pb2)
3. No. single directory.
4. when i try to run client.py with python2  ,then same error messsage came up.

Todspaniard, 2018-04-03:
actual situation is following. Sorry i' novice  on this area 
>>> import carla
>>> from carla import carla_server_pb2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/xxxx/carla/PythonClient/carla/carla_server_pb2.py", line 9, in <module>
    from google.protobuf import symbol_database as _symbol_database
  File "/home/xxxx/.local/lib/python2.7/site-packages/google/protobuf/symbol_database.py", line 61, in <module>
    from google.protobuf import descriptor_pool
  File "/home/xxxxx/.local/lib/python2.7/site-packages/google/protobuf/descriptor_pool.py", line 67, in <module>
    _USE_C_DESCRIPTORS = descriptor._USE_C_DESCRIPTORS  # pylint: disable=protected-access
AttributeError: 'module' object has no attribute '_USE_C_DESCRIPTORS'

laurent-george, 2018-04-04:
try  

  sudo apt-get remove python-protobuf
  sudo pip install --upgrade protobuf ?

and retry import carla

Todspaniard, 2018-04-06:
thanks  a lot.
Finally I got it and working well.

closed 2018-04-06 by Todspaniard

wolfsubak, 2018-08-21:
carla was working well until I tried to setup carla_ros_bridge.
OS: `Ubuntu 16.04`
Python: `2.7`
ros: `kinetic`
UE: `4.19`
carla: `0.9.0-35`

I have followed latest doc `building on linux` then followed the `Readme.md` related to ros_bridge.

Having my server first:
`./CarlaUE4.sh -windowed -ResX=640 -ResY=480`

Then I tried:
`rostest carla_ros_bridge ros_bridge_client.test`

that is resulting to a failure:

```
[ROSUNIT] Outputting test results to /home/wolf/.ros/test_results/carla_ros_bridge/rostest-test_ros_bridge_client.xml
[INFO] [1534819487.106891]: Trying to connect to localhost:2000
[INFO] [1534819487.107187]: Connected
[INFO] [1534819487.107309]: Starting Episode --> 0
[INFO] [1534819487.108934]: Adding sensor camera_front
[INFO] [1534819487.109119]: Adding sensor camera_depth
[INFO] [1534819502.124726]: Exiting Bridge
Traceback (most recent call last):
  File "/home/wolf/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/client.py", line 42, in <module>
    main()
  File "/home/wolf/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/client.py", line 38, in main
    carla_ros_bridge.run()
  File "/home/wolf/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 137, in run
    scene = self.client.load_settings(self.carla_settings)
  File "/home/wolf/carla/PythonClient/carla/client.py", line 75, in load_settings
    return self._request_new_episode(carla_settings)
  File "/home/wolf/carla/PythonClient/carla/client.py", line 160, in _request_new_episode
    data = self._world_client.read()
  File "/home/wolf/carla/PythonClient/carla/tcp.py", line 73, in read
    header = self._read_n(4)
  File "/home/wolf/carla/PythonClient/carla/tcp.py", line 89, in _read_n
    self._reraise_exception_as_tcp_error('failed to read data', exception)
  File "/home/wolf/carla/PythonClient/carla/tcp.py", line 97, in _reraise_exception_as_tcp_error
    raise TCPConnectionError('%s%s: %s' % (self._logprefix, message, exception))
carla.tcp.TCPConnectionError: (localhost:2000) failed to read data: timed out
[INFO] [1534819502.125450]: Shutdown requested
[Testcase: testtestTopics] ... ERROR!
ERROR: max time [20.0s] allotted for test [testTopics] of type [rostest/publishtest]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

rostest log file is in /home/wolf/.ros/log/rostest-BALTHAZAR-8746.log
```

As I said since I've set up carla, the clients does not work anymore:

```
wolf@BALTHAZAR:~$ python ~/carla/PythonClient/client_example.py 
INFO: listening to server localhost:2000
CarlaClient connected
ERROR: (localhost:2000) failed to read data: timed out
CarlaClient connected
^C
Cancelled by user. Bye!
```

```
wolf@BALTHAZAR:~$ python ~/carla/Dist/0.9.0-35-gc4381df/LinuxNoEditor/example.py 
Traceback (most recent call last):
  File "/home/wolf/carla/Dist/0.9.0-35-gc4381df/LinuxNoEditor/example.py", line 114, in <module>
    main(add_a_camera=False, enable_autopilot=True)
  File "/home/wolf/carla/Dist/0.9.0-35-gc4381df/LinuxNoEditor/example.py", line 43, in main
    client = carla.Client('localhost', 2000)
AttributeError: 'module' object has no attribute 'Client'

```

wolfsubak, 2018-08-21:
Then it makes more sens, I will build 0.8.4 (stable), I just finished to build UE 4.18 as well.

Thanks for the answer.

Rooparun, 2019-02-21:
Hi, 

I am new to ROS and Carla. I am getting this error. Could you please advice. 

Running the roslaunch brings up the following msg 
$ roslaunch carla_ros_bridge client.launch

Ubuntu 16.04 LTS
Carla 0.8.2
ROS Lunar
Python 2.7

PARAMETERS
 * /carla/Episodes: 2
 * /carla/Framesperepisode: -1
 * /carla/NumberOfPedestrians: 10
 * /carla/NumberOfVehicles: 30
 * /carla/QualityLevel: Low
 * /carla/SendNonPlayerAgentsInfo: True
 * /carla/SynchronousMode: True
 * /carla/WeatherId: 1
 * /carla/host: localhost
 * /carla/port: 2000
 * /carla/sensors/camera_depth/SensorType: CAMERA
 * /carla/sensors/camera_depth/carla_settings/FOV: 90
 * /carla/sensors/camera_depth/carla_settings/ImageSizeX: 320
 * /carla/sensors/camera_depth/carla_settings/ImageSizeY: 240
 * /carla/sensors/camera_depth/carla_settings/PositionX: 1.8
 * /carla/sensors/camera_depth/carla_settings/PositionY: 0
 * /carla/sensors/camera_depth/carla_settings/PositionZ: 1.3
 * /carla/sensors/camera_depth/carla_settings/PostProcessing: Depth
 * /carla/sensors/camera_depth/carla_settings/RotationPitch: 0
 * /carla/sensors/camera_depth/carla_settings/RotationRoll: 0
 * /carla/sensors/camera_depth/carla_settings/RotationYaw: 0
 * /carla/sensors/camera_front/SensorType: CAMERA
 * /carla/sensors/camera_front/carla_settings/FOV: 90
 * /carla/sensors/camera_front/carla_settings/ImageSizeX: 320
 * /carla/sensors/camera_front/carla_settings/ImageSizeY: 240
 * /carla/sensors/camera_front/carla_settings/PositionX: 1.8
 * /carla/sensors/camera_front/carla_settings/PositionY: 0
 * /carla/sensors/camera_front/carla_settings/PositionZ: 1.3
 * /carla/sensors/camera_front/carla_settings/PostProcessing: SceneFinal
 * /carla/sensors/camera_front/carla_settings/RotationPitch: 0
 * /carla/sensors/camera_front/carla_settings/RotationRoll: 0
 * /carla/sensors/camera_front/carla_settings/RotationYaw: 0
 * /carla_autopilot: True
 * /curr_episode: 
 * /rosbag_fname: 
 * /rosdistro: lunar
 * /rosversion: 1.13.7

NODES
  /
    carla_ros_bridge (carla_ros_bridge/client.py)

 [1mROS_MASTER_URI=http://192.168.0.122:11311/ [0m
 ]2;/home/XXXXX/catkin_ws/src/carla_ros_bridge/client.launch http://192.168.0.122:11311/
 [1mprocess[carla_ros_bridge-1]: started with pid [7473] [0m
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
 [1mdone [0m