carla-simulator/carla #327: Does Carla use Lidar Sensor with [x, y, z, intensity] fields?

tigerk0430, 2018-04-02:
Hello :-)

I want to use this simulator in research purpose then I just need Lidar pointcloud data with [x, y, z, intensity].
I've tried to find several simulator which support this [x,y,z,intensity] fields but I haven't yet.

or If someone can append this feature in carla, I'm willing to do with them. Thanks :-)

nsubiron, 2018-04-04:
Hi @tigerk0430,

No, we only generate 3D points without intensity. Since this is a simulated Lidar all the points are perfect, all of them would have same intensity. It seems to me this can be easily simulated in Python in the client-side, just adding an extra layer on top of our API.

tigerk0430, 2018-04-04:
@nsubiron  
got it.

Thank you for the answering :-)

closed 2018-04-07 by tigerk0430

Ajk4, 2019-01-15:
@nsubiron 

Just adding my 2 cents :D 

Intensity is not only about 'perfection' of a point. It's also about reflectivity of a surface.

Some surfaces are more reflective - for example lane markings are more reflective than concrete and lasers hitting them have higher intensity.

I would imagine that taking advantage of 'lane markings are more reflective' signal via intensity could be relevant in autonomous driving.

analog-cbarber, 2019-01-15:
In a real LIDAR system the points are just estimates. The intensity is a function of the material properties of the surfaces, their orientations with respect to the LIDAR sensor, as well as distance and atmospheric conditions. Generally a lower intensity implies the estimate will be less reliable. So it is true that a "perfect" LIDAR signal would be expected to have high intensity for all points, but would also not be realistic. 

Computing a realistic intensity function would require the same type of computations that go into a regular shader, so this should be feasible but is a lot more complicated than simply looking up the material properties of the scene. Another issue is be that the material properties in Unreal are with respect to visible frequencies of light and may not be the same as the infrared properties.

S-kewen, 2022-10-28:
> > In a real LIDAR system the points are just estimates. The intensity is a function of the material properties of the surfaces, their orientations with respect to the LIDAR sensor, as well as distance and atmospheric conditions. Generally a lower intensity implies the estimate will be less reliable. So it is true that a "perfect" LIDAR signal would be expected to have high intensity for all points, but would also not be realistic.
> > Computing a realistic intensity function would require the same type of computations that go into a regular shader, so this should be feasible but is a lot more complicated than simply looking up the material properties of the scene. Another issue is be that the material properties in Unreal are with respect to visible frequencies of light and may not be the same as the infrared properties.
> 
> Hi, I saw that in carla 0.9.10 the intensity value for the raycast lidar is present but it is not present for the semantic lidar, by chance are you aware of how to obtain this intensity that Carla 'simulates' also for the semantic lidar?

I have the same need, have you solved it?

AitorIglesias, 2023-11-21:
> Hi, I saw that in carla 0.9.10 the intensity value for the raycast lidar is present but it is not present for the semantic lidar, by chance are you aware of how to obtain this intensity that Carla [...]

@DavideCoppola97, @S-kewen, @Paolo08 

Hi! For some reason, the semantic LiDAR does not provide intensity as normal Lidar. Here I want to remark that the intensity values provided by the LiDAR sensor are actually the intensity loss based on the atmosphere attenuation ratio. The calc of the intensity loss is made in the [ComputeIntensity](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/RayCastLidar.cpp#L93) function. The semantic LiDAR does not have the atmosphere attenuation ratio attribute so it can not be calculated. But you can calculate it in the client.

```python
attenuation_rate=0.004
    
# Read Lidar Data
semantic_lidar_dtype = np.dtype([
    ('x', np.float32), ('y', np.float32), ('z', np.float32),
    ('CosAngle', np.float32), ('ObjIdx', np.uint32), ('ObjTag', np.uint32)])
# lidar_data is a carla.SemanticLidarMeasurement object
data = np.frombuffer(lidar_data.raw_data, dtype=semantic_lidar_dtype)
    
# Get intensity loss
dist = np.sqrt(data['x']**2 + data["y"]**2 + data["z"]**2)
intensity_loss = np.exp(-attenuation_rate * dist)
```

But this value is not a realistic intensity if you want to implement a realistic intensity check the #3826 issue.

Paolo08, 2023-11-23:
Hi @AitorIglesias,
Thanks for your answer! 😄