carla-simulator/carla #328: OSError: [Errno 101] Network is unreachable

adaxidedakaonang, 2018-04-02:
When I run ./Setup.sh, I met this question after it display "Backing up existing Content...".the question is:

Backing up existing Content...
'/home/server109/ADAXI/carla-0.8.0/Unreal/CarlaUE4/Content' -> '/home/server109/ADAXI/carla-0.8.0/Unreal/CarlaUE4/Content_20180402124416'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 137, in _new_conn
    (self.host, self.port), self.timeout, **extra_kw)
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 91, in create_connection
    raise err
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 81, in create_connection
    sock.connect(sa)
OSError: [Errno 101] Network is unreachable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 560, in urlopen
    body=body, headers=headers)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 346, in _make_request
    self._validate_conn(conn)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 787, in _validate_conn
    conn.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 217, in connect
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 146, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
requests.packages.urllib3.exceptions.NewConnectionError: <requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x7f98d3243908>: Failed to establish a new connection: [Errno 101] Network is unreachable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 376, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 610, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 273, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
requests.packages.urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='docs.google.com', port=443): Max retries exceeded with url: /uc?export=download&id=1tA-WwKA0RNhBoENYunqF3-F_wg11dL86 (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x7f98d3243908>: Failed to establish a new connection: [Errno 101] Network is unreachable',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./Util/download_from_gdrive.py", line 82, in <module>
    download_file_from_google_drive(args.id, args.destination)
  File "./Util/download_from_gdrive.py", line 59, in download_file_from_google_drive
    response = session.get(url, params={'id': id}, stream=True)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 480, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 468, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 576, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 437, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='docs.google.com', port=443): Max retries exceeded with url: /uc?export=download&id=1tA-WwKA0RNhBoENYunqF3-F_wg11dL86 (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x7f98d3243908>: Failed to establish a new connection: [Errno 101] Network is unreachable',))

I guess the case is that I can not link to docs.google.com,  however I can open it in my browser, please tell me what the problam is?

nsubiron, 2018-04-04:
It seems like python is not able to connect to the network, can it be a firewall is blocking it?

As a workaround, you can skip the download in the setup step with `./Setup.sh --skip-download` and download the assets manually.

TJUsym, 2018-04-09:
I had skip the download in the setup step with ./Setup.sh --skip-download but How to download the assets manually?

nsubiron, 2018-04-09:
In the output it should give the link to the latest package of the assets. You can get it too by running "Update.sh" with the skip download flag (internally Setup calls this script)
```
$ ./Update.sh --skip-download
Skipping 'Content' update. Please manually download the package from

  https://drive.google.com/open?id=1su5wQH7KJphAcirMA8VEDLndGigQ9tzq

and extract it under Unreal/CarlaUE4/Content.
```
That's the link for the version currently at master https://drive.google.com/open?id=1su5wQH7KJphAcirMA8VEDLndGigQ9tzq