carla-simulator/carla #330: Low mode camera over-exposed

pdrews, 2018-04-02:
I am running Carla 0.8 release on Ubuntu 16.04, rendering on a TitanXp.  When adding an external camera in Low mode, the image looks very over-exposed (colors are washed out, does not match the game window render).  This in not the case when rendering in Epic mode.  Below are examples of Low mode and Epic mode comparisons, along with the code I use to add the camera.

Low mode comparison:
[image]

Epic mode comparison:
[image]

Below is the code I use to initialize the camera:
```
camera = Camera('CameraRGB')
camera.set_image_size(640,512)
camera.set_position(0.60,0,1.60) # position of the camera relative to the car in cm
settings.add_sensor(camera)
```

juaxix, 2018-04-03:
Hello @pdrews please check out the Optimizations branch, I have modified it to change the scene capture camera source to HDRNoAlpha to avoid the problem (combination of gamma + post-fx).
Now it should looks something like this

[image]

pdrews, 2018-04-03:
This works great, thanks for the quick fix!

commit 9cb1f8bb (2018-04-03): Fix #330 : Low mode camera over-exposed because by changing the CaptureComponent2D->CaptureSource to HDR without alpha
PR #333 (2018-04-13): Optimizations
Some fixes included in this pull request.

  - [x] Walkers uses a closer angle to detect vehicles and the algorithm is changed (will be moved again to async find path)
  - [x] Changes in code formatting
 
#### Description

Most of the changes are related to settings, pedestrian AIs and vehicle spawning

Fixed #246
Fixed #318
Fixed #330 
Fixed #334

  - [x] Minor bugs fixed
    - [x] Vehicles spawner spawns correctly the number specified in the configuration (carla settings file) using a timer to spawn more if there is no space at the start
    - [x] Changes in the screen captures [... cut]
closed 2018-04-13 by nsubiron