carla-simulator/carla #337: How to deeply control world generation?

wzhAptiv, 2018-04-09:
I'd like to change the type ego-car, to any other built-in car model, e.g. Volkswagen van. 
And I'd also like to control the agent car generation, e.g. only have vans in the world, or only have Fiat-500-like small cars. 

How could I make it? I only found I could control via a single random seed in python, which is not actually a 'control'. 
Should I change the Engine code, or it's possible in pure python script?

Thank you!

nsubiron, 2018-04-09:
You can change the player vehicle in the settings file (or in `CarlaSettings` class in Python)
https://github.com/carla-simulator/carla/blob/15213afea982edc5118f23e64dca91225a4c1d7b/Docs/Example.CarlaSettings.ini#L33-L36

In the future we will extend the Python API to have a better control of the world generation, but at this point to change non-player vehicles you would need to modify the Unreal project.

Open the blueprint "VehicleSpawner"
[image]

Go to the function "SpawnVehicle" and select the variable "Vehicle Models"
[image]

This is the list of vehicles that is spawned into the world, you can add or remove elements here. E.g. if you remove all but the Volkswagen T2, only vans will be spawned.

wzhAptiv, 2018-04-09:
@nsubiron Thank you very much for the help!

wzhAptiv, 2018-04-16:
Hello @nsubiron , since my hardware is not powerful enough to have a high fps during simulation. Is it possible to control the world speed, e.g. limit all agent velocities to their half, so that I can assume the frame rate as doubled, using the same hardware. 

Thank you!

wzhAptiv, 2018-04-16:
@nsubiron Thank you for your explanation. But according to this, I think it's still not able to slow down the world, because -fps setting only affect the interval of re-rendering, but not affecting to the game time. 

Assume that I have a car running at 10m/s, starting at 0m
if I have -fps=5, each frame, I can get the car at [2m, 4m, 6m, 8m, 10m]
if I have -fps=10, each frame, I can get the car at [1m, 2m, 3m, ..., 8m, 9m, 10m]

What I wanted to have is that, 
if I have a 1s in game = 2s in world, I have -fps=5, I can get the car at [1m, 2m, 3m, 4m, 5m], then in the 2nd second, I get the car at [6m, 7m, 8m, 9m, 10m]. 
So although I run two seconds, I receive 10 frames, they are equal to a 10fps simulation, if I take the frame interval as 1/10 second. 

I don't know if I'm understanding your explanation correctly. But I think there should be a baseline frame rate, so that -fps can work as I had expected. 
E,g, The game always take 30 re-rendering as one second in game time, If I only make 10 re-rendering (10fps), it will only be 1/3 second in the game time, but already 1 second in our world time. So if I run it for 3 seconds in our world time, it will be 30 re-renderings in the game, which is thought to be only 1 second by the game engine. 

Sorry for my slowness in understanding, but I still don't get it. 
Thank you for your help!

nsubiron, 2018-04-16:
If I understood, what you want is sort of 2x speed setting? If that's the case we don't have that functionality. The closer think is the fixed frame rate mode.

> if I have -fps=5, each frame, I can get the car at [2m, 4m, 6m, 8m, 10m]
> if I have -fps=10, each frame, I can get the car at [1m, 2m, 3m, ..., 8m, 9m, 10m]

This is exactly how it works. Taking into account that each frame takes approximately the same amount of real-time to be computed, the `-fps=10` takes about twice to simulate 1 second.

wzhAptiv, 2018-04-16:
You're right. I want a 0.5x speed setting for the whole world, because I want to simulate the camera (e.g. 25-30fps) but my hardware can only run at 5fps, So the only way to make everything as is, is to slow down the game world. I tried to find a way to limit the agent vehicle speed within python script, but I didn't find the approach to do that. 
I feel it could be great if we could run the game world in 'a slow motion', so we can have a very high fps even if our hardware is not capable for that.

nsubiron, 2018-04-16:
But isn't that what the fixed frame-rate does? If you run the simulator with

    ./CarlaUE4.sh -benchmark -fps=60

it simulates 60FPS in slow motion.

wzhAptiv, 2018-04-16:
You mean if I set -fps=60, but my hardware can only run around 5fps, it will be that in 12 seconds, I can receive 60 frames which are only one second in game world? 
Because normally, we extract the sensor data from a python client. I'm not sure if the python client can have different frame as the server has, which means it can retrieve data from some previous time step but not current. 

I feel that every-time I call 'read_data' by the client, it will then render the current frame based on the time the server receives this call. So I'm still confused with the matching/conversion of game time with world time, when my hardware cannot reach whatever I set to -fps.

nsubiron, 2018-04-16:
> You mean if I set -fps=60, but my hardware can only run around 5fps, it will be that in 12 seconds, I can receive 60 frames which are only one second in game world?

Yes, exactly.

Now, the client-server synchronization is a different thing (and sorry, not yet well documented as you see in the "todo" of the link I sent you before). The synchronization is not done at the `read_data` step, but at `send_control` step.

You can run in _synchronous_ or _asynchronous_ modes, this is selected in the "CarlaSettings" at the beginning of each episode. This controls whether the simulator waits for the control message of the client or not.

  * In asynchronous mode, the simulator does not wait for the client, if the control message is not received the server doesn't change the state of the vehicle.
  * In synchronous mode, the simulation is halted until the control message is received.

Synchronous mode is useful if your Python client is slower than the simulator and you cannot miss measurements and control. On the other hand, async mode is slightly faster since it doesn't have to halt execution; but, if your client is not able to read the images and measurements on time, those are overwritten by the images and measurements of the next frame.

If you run in synchronous mode it is very important to use the fixed frame-rate too, otherwise the physics engine can get a bit crazy when you halt it.

wzhAptiv, 2018-04-16:
Thank you very much for the explanation. I think I used sync mode. But I'll have to try it tomorrow, if I set -fps=30, and a relative slow python client loop (I added time.sleep() to keep my python having a stable frame rate). I'd expect to see a 'slow-motion' effect. 
If I have any further problem, I would ask then. Anyway, thank you very much for your time on my questions.

nsubiron, 2018-04-16:
No worries :+1: the combination of fixed frame-rate and synchronous mode should work for slow client, slow server, or both.

closed 2018-04-30 by nsubiron