carla-simulator/carla #338: Changing Sun Azimuth angle in CarlaWeather.ini has no effect

simmates, 2018-04-09:
I'm trying to change the sun position to get shadows at different angles, however changing in CarlaWeather.ini SunAzimuthAngle=40.0
has no effect. Changing SunPolarAngle=somevalue works though. I'm using a modified level as suggested in the map customization tutorial.
Can anybody confirm this? I also tried changing it via the weather blueprint but the direction of the sunlight is always the same.

simmates, 2018-04-09:
Changing the Rotation of 'Light Source' by hand is kind of a workaround.

TheNihilisticRobot, 2018-04-10:
Hi @simmates. How our weather system works now you can not change the azimuth angle from the CarlaWeather.ini file. We pretended that this would make the azimuth angle native to the level, but it was not the best approach. I'm marking it as an bug and it will be fixed in the next version. We will make the azimuthal angle of CarlaWeather.ini be added to the position of the local sun.

PR #343 (2018-04-11): Updated weather configuration
#### Description

Made the sun azimuth editable from the CarlaWeadther.ini files

Fixes #338
closed 2018-04-11 by nsubiron