carla-simulator/carla #339: How to control differential wheeled vehicle?

fmazzi, 2018-04-09:
Hi, I wish I could use carla with differential wheeled vehicles (like skid-steer).
I'm not very experienced in the Unreal Engine and I do not know where to start.
I know the client sends a control packet to the server every frame. I would like to use the "steer" value to turn the vehicle or directly control the two wheels.
It is possible to make this modification only on the model of the vehicle in the unreal engine or I must also modify the C ++ classes of Carla? Do you have any suggestions? where can I start from?
Thanks for your help.

nsubiron, 2018-04-10:
Hi @fmazzi,

At first sight it doesn't seem an easy task. CARLA uses the PhysX vehicle model implemented in [UWheeledVehicleMovementComponent][movcomp], I couldn't find anywhere a way of making this model work with differential steering.

Probably the best way to go is to implement a different type of vehicle. You can look for how tanks are implemented in Unreal, e.g. [Blueprint Tank][bptank]. Also another place worth taking a look is at [PhysX tank vehicle][pxtank].

However, you most probably need to modify our C++ code, either modify [ACarlaWheeledVehicle][cwv] or make a similar class that can replace this one.

Let us know if you make any progress or you find more specific issues.

Good luck!

[movcomp]: http://api.unrealengine.com/INT/API/Plugins/PhysXVehicles/UWheeledVehicleMovementComponent/index.html
[bptank]: https://docs.unrealengine.com/en-us/Videos/PLZlv_N0_O1ga0aV9jVqJgog0VWz1cLL5f/Oh7pDL8RFaI
[pxtank]: http://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/apireference/files/classPxVehicleDriveTank.html
[cwv]: https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.h

closed 2018-04-30 by nsubiron

bradsquicciarini, 2020-03-05:
@fmazzi did you ever end up implementing this?

fmazzi, 2020-03-06:
Hi @bradsquicciarini 
i'm sorry I can't be very useful to you because it's a old project that then I abandoned.
The only thing I can tell you is that I had achieved some results by looking at this plugin: 
[https://www.unrealengine.com/marketplace/en-US/product/vehicles-movement-components](url)
Creating a new WheeledVehicleMovementComponentTank.

Good luck!

bradsquicciarini, 2020-03-06:
@fmazzi I've decided to use IsaacSim, but thank you for the response.