carla-simulator/carla #345: Car sensors position not relative to car center

marcgpuig, 2018-04-12:
The sensor position are relative to the car pivot (at the road), not to the center of the bounding box. Now the server sends the agent `transform.position` as the center position of the bounding box. 

Due to this, some transformations the we have in our python Carla module are now incorrect.

JulianoLagana, 2018-04-17:
Hey @marcgpuig.

Unfortunately, reading the provided link didn't answer the question. I do however understand that this is beyond what CARLA is in charge of, so I'll continue my search within PhysX (I'm actually hoping the accelerations are of the pivot point, otherwise CARLA is providing me some measurements w.r.t. the pivot point, and other measurements w.r.t. the center of mass).

Thanks for the help.

LittlePower97, 2021-05-17:
> The sensor position are relative to the car pivot (at the road), not to the center of the bounding box. Now the server sends the agent `transform.position` as the center position of the bounding box.
> 
> Due to this, some transformations the we have in our python Carla module are now incorrect.

Hi @marcgpuig,

As this post is quite old, I wanted to know if these information were still accurate.

Thank you in advance.

commit 95f25531 (2018-04-16): Fix #345 rollback vehicle's location and send relative transform of the bounding box
commit d6b61aac (2018-04-17): Fix #345 rollback vehicle's location and send relative transform of the bounding box