carla-simulator/carla #346: ServerTimeOut always set to 10000 ms regardless of provided config

Ajk4, 2018-04-12:
As in a title - it's impossible to set value other-than-default via string-provided configuration.

I am using PythonClient provided by carla and I am settings the configuration via string (protobuf messages)

Regardles of what I'm doing, I always get 10000ms in Carla logs:

```
[2018.04.12-13.24.04:214][484]LogCarla: == CARLA Settings ==============================================================
[2018.04.12-13.24.04:214][484]LogCarla: Last settings file loaded: <string-provided-by-client>
[2018.04.12-13.24.04:214][484]LogCarla: [CARLA/Server]
[2018.04.12-13.24.04:214][484]LogCarla: Networking = Enabled
[2018.04.12-13.24.04:214][484]LogCarla: World Port = 39425
[2018.04.12-13.24.04:214][484]LogCarla: Server Time-out = 10000 ms
[2018.04.12-13.24.04:214][484]LogCarla: Synchronous Mode = Enabled
[2018.04.12-13.24.04:214][484]LogCarla: Send Non-Player Agents Info = Disabled
[2018.04.12-13.24.04:214][484]LogCarla: [CARLA/LevelSettings]
[2018.04.12-13.24.04:214][484]LogCarla: Player Vehicle        = Default
[2018.04.12-13.24.04:214][484]LogCarla: Number Of Vehicles    = 15
```

It's really crucial for doing any RL experiments with Carla to have larger timeout there. Any Policy Updates will very likely take longer than 10 seconds.

Perhaps we could even set up a default value to be much larger than 10 seconds?

Ajk4, 2018-04-12:
I also provided -carla-settings argument when starting server with timeout specified and it did not work either.

Ajk4, 2018-04-12:
It seems like passing absolute path doesn't work with -carla-settings parameter.

I put my config file in carla root dir and provided relative path parameter and it worked that way.

nsubiron, 2018-04-12:
> It seems like passing absolute path doesn't work with -carla-settings parameter.

Yup, that's a known issue #215.

Some server related settings only work when passed by command-line, this is done this way because we need to know these values before a client connects.

https://github.com/carla-simulator/carla/blob/6539eaec2cc0b5f280ee1c6d6cfb20de96447ffc/Docs/Example.CarlaSettings.ini#L6-L18

Ajk4, 2018-04-12:
>Some server related settings only work when passed by command-line, this is done this way because we need to know these values before a client connects.

I see. I am in charge of my Carla server so I was able to change it. However it might be the case that someone would not be ;P

Anyways my issue is resolved. I'm closing the ticket.

closed 2018-04-12 by Ajk4

nsubiron, 2018-04-12:
Great, we'll keep this in mind when implementing new changes in the server code.