carla-simulator/carla #347: Demofile playback support?

candytaco, 2018-04-12:
Hey guys!

First off, thanks for making such a great program!

Do you guys have any plans for adding support for playing back demofiles recorded using Unreal? Right now, attempting to run `demoplay` on a demofile from the Unreal console results in an assertion failure in `UAgentComponent` attempting to get a `GameMode` object from the world. `UWorld::GetAuthGameMode()` only returns a valid pointer when in server mode. I'm not familiar with Unreal, but it seems like, as in Source (which I'm more familiar with), the demofiles are recorded as the game client and played back as the client.

The use case is to have a human drive in real time (which requires running the simulator at 30+ fps in real time), recorded using Unreal. Afterwards, the file will be played back to render depth maps/lidar and semantic segmentation also at 30+ fps, but not necessarily in real time.

Cheers

juaxix, 2018-04-13:
Maybe @nsubiron can talk more about this feature, because it could be part of the next features to include

nsubiron, 2018-04-16:
@candytaco We'll most probably implement some sort of log of everything that happens in the world and a way to play back these logs. This is a feature we have in our backlog but we haven't planned yet when this will be incorporated into CARLA.

candytaco, 2018-05-26:
Hey guys,

I've been poking around and have managed to get `AWheeledVehicleController::PlayerCamera` and `AWheeledVehicleController::OnBoardCamera` to re-attach correctly to the player vehicle's pawn when playing back recorded demo files. I can currently recreate what happened in the server window (sans HUD) when a session is recorded and played back with the mock game controller. Not sure what happens with an actual server controller yet, or whether sensors can be attached.

Cheers

nsubiron, 2019-04-05:
This was added in 0.9.4.

closed 2019-04-05 by nsubiron