carla-simulator/carla #350: Exposing PhysX functionality

dpattison3, 2018-04-13:
I'm am trying to expose the full velocity/angular velocity of vehicles. I would also like to expose accurate ground truth acceleration data to get a simulated IMU sensor. I think the best way to do this is to have access to the underlying PhysX engine.
At the moment, I am accessing the PxVehicleWheels PhysX class through a call to GetVehicleMovementComponent from ACarlaWheeledVehicle. This has the full linear/angular velocities I want. However, it complains because the associated header file isn't included. I don't really understand the build system well enough to be able to modify it to get the right include files. 

Does know how I can go about getting PhysX functionality?

dpattison3, 2018-04-13:
Sorry, I should have tried for a little longer before opening an issue.
You can add PhysX and APEX to the modules list in Carla.Build.cs. Then you can include PhysXIncludes.h header file. Then you have access to the PhysX functionality.

closed 2018-04-13 by dpattison3

GeorgeMD, 2018-08-08:
Can you make a full tutorial on how to get this done, or maybe a quick step by step? I need access to this kind of information, but I am not sure how to do what you did.

dpattison3, 2018-08-09:
Not really deserving of a full tutorial. It really is as simple as added two lines to a file. In  carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs  add "PhysX", and "APEX" to the PrivateDependencyModuleNames list (ie. add them at line 44 where it tells you to add extras you need.
Now, you have the PhysX functionality and can access it according to their docs. That's it.

GeorgeMD, 2018-08-11:
Thanks a lot.

cedros23, 2018-09-18:
Hello,  thanks for the information regarding PhysX functionality. I am after IMU simulation as well and despite the efforst, I could not locate particular method or function in PxWheelVehicles class to get angular velocity. Would you please advise? Thanks in advance.

dpattison3, 2018-09-18:
Once you have access to PhysX, it's up to you to figure out what you need from it. But specifically to get angular velocities, ACarlaWheeledVehicle class in this repository, somewhere along the inheritance chain has access to a rigid dynamics class. So, this might get you what you want: GetVehicleMovementComponent()->PVehicle->getRigidDynamicActor()->getAngularVelocity()

zoeeer, 2019-01-07:
seems like I have access to the PVehicle instance but I cannot call its methods.

> So, this might get you what you want: 
> GetVehicleMovementComponent()->PVehicle->getRigidDynamicActor()->getAngularVelocity()

I got a link error compiling this code. Is there any headers I should include in my .cpp file?

zoeeer, 2019-01-07:
I found out that I needed "PhysXVehicleLib" module too.

https://answers.unrealengine.com/questions/722663/physx-linker-errors-custom-vehiclemovementcomponen.html