carla-simulator/carla #352: Pedestrians with a custom path

ciumonk, 2018-04-14:
I've been working on this for a while now. Here what works:
-spawning a pedestrian(or more) with a certain trajectory and speed, configurable from the settings file or sent from the python client(to be used in crash simulations). This basically uses a combination of SetActorLocation and Controller->MoveTo along with selectively disabling the part of code that stops the pedestrian if he is about to intersect with a vehicle.
-the same with cars

My only issue is that pedestrians seem to remain stuck around intersections sometimes. I haven't been able to find any code that would affect this. I have a suspicion that it might be somewhere in a blueprint, any ideas? Pedestrians spawned on the sidewalk in the vicinity of intersections sometimes never move using Controller->MoveTo, unless I interpolate using SetActorLocation(start+ratio*end)....

Is there something that interferes with the NavMesh if there are RoutePlanner nodes around?

ciumonk, 2018-04-14:
Ok, I see, the navmesh is changed using NavModifiers. Ok, so removing the relevant NavModifiers allows me to have pedestrians across intersections as well.

I guess you can close this, it was pretty pointless...

marcgpuig, 2018-04-16:
@ciumonk glad it worked :)

closed 2018-04-16 by marcgpuig

simmates, 2018-04-16:
@ciumonk nice work, can you provide information on how you implemented this... did you use blueprints or did you extend some of the .cpp code? Maybe it's interesting to merge this functionality into carla - I would definitely be interested.

ciumonk, 2018-04-18:
Hi, I extended the .cpp code. It's not that much really, just a new WalkerSpawnerBase and some modifications to WalkerAIController, CarlaSettings and the maps.

Perhaps I could submit a PR after the ITSC2018 deadline(1 May)...

kristinayige, 2018-05-17:
Hello, I'm very curious about how did you do it (and I'm supposed to do the same thing). Could you share some details about your modifications? Any help is appreciated!

bjane123, 2018-10-23:
Hi, @marcgpuig @ciumonk  I have a problem in changing walker spawn points. I only leave two walkerspawnpoints in the map.  I thought walkers should be moving between the two points, but it show no walker at all. I don't know why.

marcgpuig, 2018-10-24:
@TheNihilisticRobot maybe can help!

TheNihilisticRobot, 2018-10-24:
Hi @bjane123 

There's many things that can prevent Walkers from spawning. Maybe the spawn point is too close to the floor; too high in the air or to close to any object. Maybe there isn't navigation built in the level (#535)  It could also be that you are using only starter walker spawn points (Points that only spawn pedestrians but don't work as targets for other pedestrians); or that your CarlaSettings have the number of pedestrians set to 0.