carla-simulator/carla #355: Clarification regarding player measurements

JulianoLagana, 2018-04-16:
I would like to obtain some clarification regarding the player measurements available in CARLA. 

In https://carla.readthedocs.io/en/latest/measurements/, the following table is given:

Player measurements
-------------------

Key                        | Type      | Units  | Description
-------------------------- | --------- | ------ | ------------
transform                  | Transform |        | World transform of the player (contains a locations and a rotation).
acceleration               | Vector3D  | m/s^2  | Current acceleration of the player.
forward_speed              | float     | m/s    | Forward speed of the player.
collision_vehicles         | float     | kg*m/s | Collision intensity with other vehicles.
collision_pedestrians      | float     | kg*m/s | Collision intensity with pedestrians.
collision_other            | float     | kg*m/s | General collision intensity (everything else but pedestrians and vehicles).
intersection_otherlane     | float     |        | Percentage of the car invading other lanes.
intersection_offroad       | float     |        | Percentage of the car off-road.
autopilot_control          | Control   |        | Vehicle's autopilot control that would apply this frame.

Questions:
- With respect to which coordinate frame is the acceleration of the player given? The world frame, the car frame, another one?
- What is "forward speed" exactly? From the name it seems that we project the velocity vector into the line tangent to the car's trajectory at each instant, and compute the magnitude. From my experiments it seems to just be the magnitude of the velocity vector (no projections involved). Which is it?

I'd suggest adding this information to the docs.

Thanks!

marcgpuig, 2018-04-16:
Hi @JulianoLagana,

> The x-y-z coordinates in the "World transform of the player" are of which point of the car? I imagined it would be of it's center of mass (or at least its geometric center), but they seem to be from its pivot point.

The center of the bounding box, but there is an issue, take a look at #345.

> With respect to which coordinate frame is the acceleration of the player given? The world frame, the car frame, another one?

The world frame.

> What is "forward speed" exactly? From the name it seems that we project the velocity vector into the line tangent to the car's trajectory at each instant, and compute the magnitude. From my experiments it seems to just be the magnitude of the velocity vector (no projections involved). Which is it?

This is a PhysX calculation. Looking at his code, seems that is the linear speed projected to the forward vector of the chassis of the vehicle.

JulianoLagana, 2018-04-17:
@marcgpuig: Thanks a lot for the quick reply. The explanation regarding the x-y-z coordinates also clarifies a lot the point you make regarding the velocity vector (assuming no tire slip, the velocity at the pivot point will always be parallel to the forward vector of the chassis of the vehicle). In short, your definitions explain everything I'm seeing in my code, thank you.

marcgpuig, 2018-04-17:
@JulianoLagana Great :)

Thanks for your valuable feedback!
I will close this issue when I have added this to the documentation.

closed 2018-05-30 by nsubiron

alexanderkoumis, 2019-03-28:
Hello, is `forward_speed` still available in the Python API for version 0.9.4? Or should we calculate this value ourselves given `actor.get_transform().get_forward_vector()` and `actor.get_velocity()`?

nsubiron, 2019-03-28:
@alexanderkoumis Yes, `forward_speed` was replaced by the more general `velocity` vector, now you need to calculate the forward speed yourself.

mitulbhesaniya, 2019-11-26:
@nsubiron  Hi, There is read_data() in the stable version for getting the player_measurements. I want to calculate the orientation which is in measurements.player_measurements.transform.orientation for stable version. so I want to get this information for latest version. How can I do that and how it is being replaced in latest version? Please help me if you have any idea. 
Thank you!

KShah707, 2020-02-19:
> Hello, is `forward_speed` still available in the Python API for version 0.9.4? Or should we calculate this value ourselves given `actor.get_transform().get_forward_vector()` and `actor.get_velocity()`?
> 
> Edit: Like this:
> 
> ```
> yaw_global = np.radians(vehicle.get_transform().rotation)
> 
> rotation_global = np.array([
>     [np.sin(yaw_global),  np.cos(yaw_global)],
>     [np.cos(yaw_global), -np.sin(yaw_global)]
> ])
> 
> velocity_global = vehicle.get_velocity()
> velocity_global = np.array([velocity_global.y, velocity_global.x])
> velocity_local = rotation_global.T @ velocity_global
> ```

For future readers, note that the Transform object now has a `get_forward_vector()` method to make this a little easier

commit 3850f39d (2018-05-29): Added certain measurements clarifications. This fixes #355