carla-simulator/carla #357: Help requested: Carla 0.8.1 crash on assets ?

gilamid, 2018-04-17:
Downloaded and extracted CARLA 0.8.1 on Ubunto 16. 

Carla server is crashing with:

2018.04.17-12.33.48:823][  0]LogInit: Display: Game Engine Initialized.
[2018.04.17-12.33.48:823][  0]LogGameplayTags: Display: UGameplayTagsManager::DoneAddingNativeTags. DelegateIsBound: 0
[2018.04.17-12.33.48:921][  0]LogInit: Display: Starting Game.
[2018.04.17-12.33.48:921][  0]LogNet: Browse: /Game/Maps/Town01?Name=Player
[2018.04.17-12.33.48:922][  0]LogLoad: LoadMap: /Game/Maps/Town01?Name=Player
[2018.04.17-12.33.52:228][  0]LogStreaming: Error: ****DumpDependencies [Dependencies]:
[2018.04.17-12.33.52:229][  0]LogStreaming: Error:     Linker is ../../../CarlaUE4/Content/Static/Buildings/Bl_CityBuilding_SupermarketWOffice/SupermarketWOfficeN2.uasset
[2018.04.17-12.33.52:229][  0]LogStreaming: Error:         Dep S_BEFORE_S Import     5   /Script/Engine.Actor
[2018.04.17-12.33.52:229][  0]LogStreaming: Error:         Dep S_BEFORE_S Import     0   /Script/Engine.Default__Actor
[2018.04.17-12.33.52:229][  0]LogStreaming: Error:         Dep S_BEFORE_C Import     6   /Script/Engine.BlueprintGeneratedClass
[2018.04.17-12.33.52:229][  0]LogStreaming: Error:         Dep S_BEFORE_C Import     1   /Script/Engine.Default__BlueprintGeneratedClass
[2018.04.17-12.33.52:229][  0]LogStreaming: Error:         Dep C_BEFORE_C Import     5   /Script/Engine.Actor
Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 

Seems like an assets issue ? Any help ?

Thanks!

marcgpuig, 2018-04-17:
Hello @gilamid 

We haven't seen this problem before, can you provide us more information?
Which Unreal version are you using? How about your hardware?

gilamid, 2018-04-17:
Hi, 

I see this on Carla 0.8.1 and also on 0.8.0

I am using a g2.2Xlarge AWS instance , ubuntu 16. 

(The g2.2xlarge version comes with 15 GiB memory, 60 GB of local storage, 26 EC2 Compute Units (that’s an Intel Sandy Bridge processor running at 2.6 GHz) and a single NVIDIA  <http://www.nordichardware.com/Graphics/nvidia-kepler-gk104-gets-1536-cuda-cores.html> Kepler GK104graphics card (with 1536 CUDA cores).)

The only thing I can add is :

Carla itself is not installed on local disk of the instance, but rather on an NFS mounted volume – from another instance. 

 

I can add the following info – the first warning I can detect in the log file is :

LogFileManager: Warning: ReadFile failed: Count=0 Length=4096 Error=errno=21 (Is a directory) for file 

 

 

I am also attaching a full log file. 

Thanks for your help.

Gil

gilamid, 2018-04-18:
> Hey @gilamid <https://github.com/gilamid> ,
>
> I found a copy of SupermarketWOfficeN2.uasset called SupermarketWOfficeN2ç.uasset (notice the final ç). This can be causing some troubles. I asked the artists to change this on the main branch, but meanwhile I can send you a correct copy of that specific folder for you to test it.
>
> I can send it to your email, shown on your previous comment, if you agree.
>
> Regards.

Please do !

Thanks,

Gil

gilamid, 2018-04-20:
Update: I did nto install the asset that you sent me. BUT - I did create another AWS instance, and went carefully on the drivers and VGLrun install. The exact same executable that crashed on the previous AWS instance - is now running cleanly. So , obviously something in my environment triggered this crash, but as I am having a running version - I will move debugging it to a lower priority.

closed 2018-05-16 by marcgpuig