carla-simulator/carla #359: Editing asset in release version

alexandru-dinu, 2018-04-17:
Hello!

My goal is to edit some parameters of the Mustang asset and be able use the altered vehicle in the release version of Carla. 
I have made the edits in UE4 and now I want to use the altered files in the release. Because the release uses the `CarlaUE4-LinuxNoEditor.pak` archive, I have tried a little _workaround_, that is to unpack it (using [2]), replace the car's files and pack it again. Unfortunately, it doesn't work as expected, throwing some errors [3].

Now, using [2] was clearly not the _right_ way to achieve this (the developer himself said that it wasn't tested for UE4, my guess is that headers / offsets differ), so apart from rebuilding the whole thing is there any way to do this?

Thanks!

[1] `Mustang_FrontWheel.uasset, Mustang_RearWheel.uasset, Mustang.uasset`
[2] https://github.com/panzi/u4pak
[3] `Preparing to exit.Shutting down and abandoning module NetworkFile (10)Shutting down and abandoning module CookedIterativeFile (8)Shutting down and abandoning module StreamingFile (6)Shutting down and abandoning module SandboxFile (4)Shutting down and abandoning module PakFile (2)Exiting.Exiting abnormally (error code: 1)`

ciumonk, 2018-04-18:
Not really, I also tried u4pak a while ago with no success. Why not edit using the Editor and then repack?

alexandru-dinu, 2018-04-18:
I wanted to see if there's a shortcut, but that's what I will do. Thanks!

closed 2018-04-18 by alexandru-dinu