carla-simulator/carla #360: Build error with 0.8.1

sandman, 2018-04-18:
Hi,
I get a build error when I run the Rebuild.sh script for Carla 0.8.1. I suspect the error might have been because I ran Setup.sh before building UnrealEngine 4.18..?

Output of ./setup.sh: https://gist.github.com/sandman/8671b041a988e01441aba1754a21e0f1

Output of Rebuild: https://gist.github.com/sandman/0533120716229d17dc2ef9918586d8bf

There seems to be an issue with linking the python headers. Any ideas how to resolve this?

Thanks,
Sandip

nsubiron, 2018-04-30:
Hi,
It seems a problem with your python installation, maybe you can try reinstalling the `python-dev` package.

tomchen1000, 2018-05-07:
https://github.com/carla-simulator/carla/issues/199

closed 2018-05-22 by nsubiron