carla-simulator/carla #361: Error while packing project

alexandru-dinu, 2018-04-18:
As mentioned in #359 , my goal is to edit Mustang asset and because the [u4pak](https://github.com/panzi/u4pak) tool is not compatible with UE4, I have to repack the whole project (generating a new pak file). Unfortunately, I am getting this error:

```
ERROR: Command failed (Result:5): 
~/UnrealEngine_4.18/Engine/Binaries/DotNET/UnrealBuildTool.exe CarlaUE4 Linux Development -Project=/home/nemodrive/carla/Unreal/CarlaUE4/CarlaUE4.uproject  ~/carla/Unreal/CarlaUE4/CarlaUE4.uproject -NoUBTMakefiles  -remoteini="~/carla/Unreal/CarlaUE4" -skipdeploy -noxge -NoHotReload -ignorejunk.
AutomationTool exiting with ExitCode=5 (5)
[2018.04.18-15.57.49:470][353]UATHelper: Packaging (Linux): RunUAT ERROR: AutomationTool was unable to run successfully.
[2018.04.18-15.57.49:470][353]PackagingResults: Error: Unknown Error
```
System info: `Ubuntu 16.04.4 LTS x64 / Linux 4.13.0-36-generic`

Is there something I am missing? 

Thanks.

alexandru-dinu, 2018-04-19:
I have managed to completely rebuild the project and construct the `.pak` file by firstly cleaning the project.

Right now, I am facing another problem when running this custom release [1] (from `LinuxNoEditor` folder: `./CarlaUE4.sh /Game/Maps/Town01 -carla-server -windowed -ResX=640 -ResY=480`). 
There is an assertion error in `sensor.py` when the client reads the data:  `assert len(raw_data) == 4 * width * height`.  

It appears that the width and height from `data` (where `data = self._stream_client.read()`) are 1, 0 respectively. There are no problems when running the official carla release (0.8.1). Is there a setting I am omitting?

[1] custom = modified some car parameters in the Mustang asset

Thanks!

alexandru-dinu, 2018-04-19:
I think I have found one source of the problem. When data is read from the stream_client: `data = self._stream_client.read()`, it is packed differently in my custom build vs in the original release (i.e.: my custom build (server) adds extra bytes after the sensor_id)

Thus, if I change the data from the stream client () to be `data = data[0:4] + data[12:]`, then the parsing works as expected and I am able to see image data.

Any ideas for where those extra 8 bytes might come from?

juaxix, 2018-04-19:
Hello @alexandru-dinu we had a similar problem recently, take a look to the issue #318 ,it only happened with Windows and Direct3D, and when the source stride is different than the destiny stride, it was fixed, have you tried using Vulkan as renderer or ,are you changing anything related to the screen captures?

alexandru-dinu, 2018-04-19:
Hello, @juaxix. 

I am not using Vulkan, nor I am changing anything related to the screen captures.
Basically what I did was to rebuild the whole Carla (clean + build) with that altered asset (Mustang car) and redo all the steps for a server-control communication. Anything else is unchanged.

juaxix, 2018-04-20:
@alexandru-dinu I was suggesting you to try to use Vulkan hehe, but ok, no worries, what I need to analyze the error, is the commands you are using for server (simulator) and python client. And some logs should be very useful too

alexandru-dinu, 2018-04-21:
@juaxix - Here are the logs for the client and the server [1]. 
The client was started as: `python manual_control.py`
The server was started as: `./CarlaUE4.sh -carla-server -windowed -ResX=640 -ResY=480`

The exact setup works without problem in Carla release 0.8.1 so there's certainly a problem with my custom build.

[1] https://gist.github.com/alexandru-dinu/2619012a4b28ec5c22499db3abb17445

Thank you for your help!

alexandru-dinu, 2018-05-06:
Hello!

Any updates on this issue?

juaxix, 2018-05-07:
Hello @alexandru-dinu , we can't currently support modifications of the delivered paks, so I would recommend you to make your own version of CARLA compiling it from scratch using [the guide](http://carla.readthedocs.io/en/latest/how_to_build_on_windows/) , and of course, try with vulkan because it uses another path to generate the images from the screen captures, you just have to add the parameter "-vulkan" to the end of the command line when you run the simulator (server)

alexandru-dinu, 2018-05-07:
_make your own version of CARLA compiling it from scratch using the guide_
I have already done this, as mentioned in the beginning of this issue. My build and release 0.8.2 both have the same assertion error problem.

juaxix, 2018-05-07:
Oh ,sorry, I misunderstood. 
I would like to know more details about your hardware, can you post info of your gpu please?

juaxix, 2018-05-07:
> [1] custom = modified some car parameters in the Mustang asset

are you sure that those modifications in the car are working?, if the car is not spawning it will lead to errors...did you run the game with no server mode to see if the car is ok?

alexandru-dinu, 2018-05-07:
GPU info: [nVidia GTX 1050 Ti mobile](https://www.techpowerup.com/gpudb/2912/geforce-gtx-1050-ti-mobile)

Update: I have managed to get CARLA release 0.8.2 to work, it was a misunderstanding. It works well now.
Regarding my custom build, I'll perform a clean-up and rebuild the whole thing with my modifications.

I think this can be closed. If any other problems occur, I'll keep you updated.

Thank you very much for your help!

closed 2018-05-07 by alexandru-dinu

Amin-Tgz, 2019-09-23:
OS Linux Ubuntu 16.04 - 64bit
Unreal 4.18.3
Carla 0.8.4
--------------------------------
I made a change in map and now want to package project for shipping mode,
how should I do that?
by rebuilding all the project? will my changes remove when executing this line or not? 
`$ UE4_ROOT=~/UnrealEngine_4.18 ./Rebuild.sh?`
using package menue in Unreal throw this error:
> Error: RunUAT.bat ERROR: AutomationTool was unable to run successfully. Cannot build.
Tnx in advance

viniciusguigo, 2020-02-14:
> For anyone has similar issues:
> I did following things:
> 
> 1. un-installed clang-5.0, make sure I am using clang-3.9 only.
> 2. Followed https://carla.readthedocs.io/en/stable/how_to_build_on_linux/
>    Remove UnrealEngine_4.18 folder and git clone it again, re-compiled it again.
> 3. Remove carla-0.8.4 branch folder and git clone it again, re-compiled it again.

Just doing step 1 (downgrading clang from 6.0 to 3.9) already solved my problem. I also have the "Full rebuild" option selected on Unreal so it makes a clean build all the time. You can change that on Project Settings > Project - Packaging. Thank you!