carla-simulator/carla #369: Pose start and end point numbering changed in new Release 0.8.1?

jcyriac2, 2018-04-21:
Referring to the poses used in corl2017.py at carla/PythonClient/carla/benchmarks/corl_2017.py (line 76), it seems that with the new 0.8.1 version of server and client, the car starts at the same start point but does not stop at the original end point like in 0.7.1. For example, with the pose [36,40] on Town1, the car is supposed to drive in a short straight path but in 0.8.1, it keeps going past [40] and takes a turn. Has there been a change in the way these start and end points are represented internally with the new release? Thanks

jcyriac2, 2018-04-21:
I believe I found the cause of this issue in "Tag 0.8.1 carla/PythonClient/carla/benchmarks/benchmark.py" (line 129,135,138).
https://github.com/carla-simulator/carla/blob/9a9ce7bfa1cc22c77e7ba36713af1ff3bd768392/PythonClient/carla/benchmarks/benchmark.py#L129-L139
 I believe the units for curr_x,curr_y was in centimeters but target.location.x,target.location.y was still in meters, so this caused line 138 to evaluate to False even when the goal was reached. Hope this helps!

felipecode, 2018-04-21:
Hey ! Yes, the benchmark was poorly adapted to 0.8 because 
I was working in a new version. Luckily it is already finished ! The PR #341 has
been merged to master. 
Check out the new documentation http://carla.readthedocs.io/en/latest/benchmark_start/

closed 2018-08-06 by stale[bot]