carla-simulator/carla #370: I can't found where ./CarlaUE4.sh is?

NozomiWyane, 2018-04-22:
I built it on Ubuntu 16.04 LTS,I could open the project following "How to build Carla on Linux",but how to run CarlaUE4.sh?I can't find it in my browser.I already run town01 and town02 on UE4Editor.But where the script is?

marcgpuig, 2018-04-23:
Hello @NozomiWyane,

This is a common question already answered several times.
Please read the [F.A.Q](http://carla.readthedocs.io/en/latest/faq/) and the issue #22.

closed 2018-04-23 by marcgpuig

kimiya66, 2019-07-25:
> Hello @NozomiWyane,
> 
> This is a common question already answered several times.
> Please read the [F.A.Q](http://carla.readthedocs.io/en/latest/faq/) and the issue #22.

and then how to access python project in this way? because when I try to run .py I got error that it could not connect to server (port 2000)