carla-simulator/carla #374: Unable to build 0.8.1 on ubuntu 16.04

YashBansod, 2018-04-23:
I tried building CARLA 0.8.1 from source in my Ubuntu machine. On running the `./Setup.sh` I get the following error:

> Folder llvm-source already exists, skipping git clone...
> -- The C compiler identification is Clang 3.9.1
> -- The CXX compiler identification is Clang 3.9.1
> -- The ASM compiler identification is Clang
> -- Found assembler: /usr/bin/clang-3.9
> -- Check for working C compiler: /usr/bin/clang-3.9
> -- Check for working C compiler: /usr/bin/clang-3.9 -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/clang++-3.9
> -- Check for working CXX compiler: /usr/bin/clang++-3.9 -- broken
> CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCXXCompiler.cmake:54 (message):
>   The C++ compiler "/usr/bin/clang++-3.9" is not able to compile a simple
>   test program.
> 
>   It fails with the following output:
> 
>    Change Dir: /home/uidk4283/carla/Util/Build/llvm-build/CMakeFiles/CMakeTmp
> 
>   
> 
>   Run Build Command:"/usr/bin/make" "cmTC_a1706/fast"
> 
>   /usr/bin/make -f CMakeFiles/cmTC_a1706.dir/build.make
>   CMakeFiles/cmTC_a1706.dir/build
> 
>   make[1]: Entering directory
>   '/home/uidk4283/carla/Util/Build/llvm-build/CMakeFiles/CMakeTmp'
> 
>   Building CXX object CMakeFiles/cmTC_a1706.dir/testCXXCompiler.cxx.o
> 
>   /usr/bin/clang++-3.9 -o CMakeFiles/cmTC_a1706.dir/testCXXCompiler.cxx.o -c
>   /home/uidk4283/carla/Util/Build/llvm-build/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
> 
> 
>   Linking CXX executable cmTC_a1706
> 
>   /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_a1706.dir/link.txt
>   --verbose=1
> 
>   /usr/bin/clang++-3.9 CMakeFiles/cmTC_a1706.dir/testCXXCompiler.cxx.o -o
>   cmTC_a1706
> 
>   /usr/bin/ld: cannot find -lstdc++
> 
>   clang: error: linker command failed with exit code 1 (use -v to see
>   invocation)
> 
>   CMakeFiles/cmTC_a1706.dir/build.make:97: recipe for target 'cmTC_a1706'
>   failed
> 
>   make[1]: *** [cmTC_a1706] Error 1
> 
>   make[1]: Leaving directory
>   '/home/uidk4283/carla/Util/Build/llvm-build/CMakeFiles/CMakeTmp'
> 
>   Makefile:126: recipe for target 'cmTC_a1706/fast' failed
> 
>   make: *** [cmTC_a1706/fast] Error 2
> 
>   
> 
>   
> 
>   CMake will not be able to correctly generate this project.
> Call Stack (most recent call first):
>   CMakeLists.txt:48 (project)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/uidk4283/carla/Util/Build/llvm-build/CMakeFiles/CMakeOutput.log".
> See also "/home/uidk4283/carla/Util/Build/llvm-build/CMakeFiles/CMakeError.log".

I have tried building it multiple times but cannot get this to work. In past I have built various versions of CARLA and never faced this problem. Can someone help me with this issue?

marcgpuig, 2018-04-23:
Hi @YashBansod,

> The C++ compiler "/usr/bin/clang++-3.9" is not able to compile a simple
test program.

The problem seems your compiler, nothing related to CARLA.
I'm sure you will find the solution on internet, seems a common problem ;)

YashBansod, 2018-04-24:
Thanks a lot @marcgpuig. I was able to resolve the issue by reinstalling gcc and g++ in ubuntu.

closed 2018-04-24 by YashBansod