carla-simulator/carla #377: Align the semantic segmentation with Cityscapes

EthanZhangYi, 2018-04-24:
Hi,
Till now only 12 categories (road, lane-marking, traffic sign, sidewalk, fence, pole, wall, building, vegetation, vehicle, pedestrian, and other.) are supplied in Carla. I am wondering if it is possible or if it is in the plan to add more categories and align the semantic classes with Cityscapes which have more categories.

marcgpuig, 2018-04-30:
Hi @EthanZhangYi,

It is entirely possible to add your own tags.
By now it is a bit complicated and requires to modify the code. It is not well documented right now but you can find a clue in the [docs](http://carla.readthedocs.io/en/latest/cameras_and_sensors/#camera-semantic-segmentation).

I'll keep this issue opened until you are able to do it.

marcgpuig, 2018-05-30:
I close this issue for lack of feedback :(

closed 2018-05-30 by marcgpuig

Nozomeister, 2018-11-09:
Hi @marcgpuig , I'm trying to align the tag ID of Carla output segmentation to the Cityscapes classes, so I just need to edit the Tagger.cpp and Tagger.h? Is that correct?

marcgpuig, 2018-11-12:
@Nozomeister yes :)