carla-simulator/carla #378: Help Requested : Cannot find server port from netstats and Errno 111

jd730, 2018-04-25:
Hi,
I am trying to test `client_example.py` However, I received below messages only although I turned on the server using `~/UnrealEngine_4.18/Engine/Binaries/Linux/UE4Editor "$PWD/CarlaUE4.uproject" -carla-server` command.

Error  : `ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused`

At first, I thought that it was firewall problem but, I found that though I turned on the server, the port is not activated according to `netstats`.

Could you help me solve this problem?

My server setting is like this

```
[2018.04.25-19.12.52:585][  0]LogCarla: == CARLA Settings ==============================================================
[2018.04.25-19.12.52:585][  0]LogCarla: Last settings file loaded: ../../../../carla/Unreal/CarlaUE4/Config/CarlaSettings.ini
[2018.04.25-19.12.52:585][  0]LogCarla: [CARLA/Server]
[2018.04.25-19.12.52:585][  0]LogCarla: Networking = Enabled
[2018.04.25-19.12.52:585][  0]LogCarla: World Port = 2000
[2018.04.25-19.12.52:585][  0]LogCarla: Server Time-out = 100000 ms
[2018.04.25-19.12.52:585][  0]LogCarla: Synchronous Mode = Disabled
[2018.04.25-19.12.52:585][  0]LogCarla: Send Non-Player Agents Info = Disabled
[2018.04.25-19.12.52:585][  0]LogCarla: [CARLA/LevelSettings]
[2018.04.25-19.12.52:585][  0]LogCarla: Player Vehicle        = Default
[2018.04.25-19.12.52:585][  0]LogCarla: Number Of Vehicles    = 15
[2018.04.25-19.12.52:585][  0]LogCarla: Number Of Pedestrians = 30
[2018.04.25-19.12.52:585][  0]LogCarla: Weather Id = 1
[2018.04.25-19.12.52:585][  0]LogCarla: Seed Vehicle Spawner = 123456789
[2018.04.25-19.12.52:585][  0]LogCarla: Seed Pedestrian Spawner = 123456789
[2018.04.25-19.12.52:585][  0]LogCarla: Found 0 available weather settings.
[2018.04.25-19.12.52:585][  0]LogCarla: [CARLA/QualitySettings]
[2018.04.25-19.12.52:585][  0]LogCarla: Quality Settings = Epic
[2018.04.25-19.12.52:585][  0]LogCarla: [CARLA/Sensor]
[2018.04.25-19.12.52:585][  0]LogCarla: Added 0 sensors.
[2018.04.25-19.12.52:585][  0]LogCarla: Semantic Segmentation = Disabled
[2018.04.25-19.12.52:585][  0]LogCarla: ================================================================================
```

Here is a server log.

```
Increasing per-process limit of core file size to infinity.
LogInit: Display: Running engine for game: CarlaUE4
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(0) SRandInit(0).
LogInit: Read version file with build ID 'b2c04a90-aebd-482b-97f0-f928baeaa875'
LogInit: Registered custom module enumerator with build ID 'b2c04a90-aebd-482b-97f0-f928baeaa875'
LogTaskGraph: Started task graph with 5 named threads and 20 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.064724
LogPluginManager: Mounting plugin PluginBrowser
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin GameplayTagsEditor
LogPluginManager: Mounting plugin SpeedTreeImporter
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioCodeSourceCodeAccess
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
[... 747 lines cut ...]
[2018.04.25-19.13.24:631][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2018.04.25-19.13.24:631][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2018.04.25-19.13.24:631][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2018.04.25-19.13.24:649][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2018.04.25-19.13.24:652][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2018.04.25-19.13.24:652][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2018.04.25-19.13.24:652][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2018.04.25-19.13.24:779][  0]LogContentBrowser: Native class hierarchy updated for 'HierarchicalLODOutliner' in 0.0005 seconds. Added 1 classes and 2 folders.
[2018.04.25-19.13.24:779][  0]LogLoad: (Engine Initialization) Total time: 46.73 seconds
[2018.04.25-19.13.24:779][  0]LogLoad: (Engine Initialization) Total Blueprint compile time: 0.00 seconds
[2018.04.25-19.13.24:779][  0]LogExternalProfiler: No external profilers were discovered.  External profiling features will not be available.
[2018.04.25-19.13.24:780][  0]LogSlate: Took 0.000087 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K)
[2018.04.25-19.13.24:783][  0]LogSlate: Took 0.000069 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Bold.ttf' (160K)
[2018.04.25-19.13.24:792][  0]LogSlate: Took 0.000086 seconds to synchronously load lazily loaded font '../../../Engine/Content/Editor/Slate/Fonts/FontAwesome.ttf' (139K)
[2018.04.25-19.13.25:015][  0]LogContentStreaming: Texture pool size now 2000 MB
[2018.04.25-19.13.25:041][  0]LogRenderer: Reallocating scene render targets to support 804x360 Format 10 NumSamples 1 (Frame:1).
[2018.04.25-19.13.25:041][  0]LogSlate: Took 0.000114 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Light.ttf' (159K)
[2018.04.25-19.13.25:478][  1]LogLinux: Setting swap interval to 'Immediate'
[2018.04.25-19.13.25:478][  1]LogLinux: Warning: Unable to set desired swap interval 'Immediate'
[2018.04.25-19.13.25:534][  1]LogLinux: Setting swap interval to 'Immediate'
[2018.04.25-19.13.25:534][  1]LogLinux: Warning: Unable to set desired swap interval 'Immediate'
[2018.04.25-19.13.26:888][ 18]LogAssetRegistry: Asset discovery search completed in 40.6796 seconds
[2018.04.25-19.13.26:913][ 18]LogCollectionManager: Rebuilt the object cache for 0 collections in 0.000001 seconds (found 0 objects)
[2018.04.25-19.13.26:913][ 18]LogCollectionManager: Fixed up redirectors for 0 collections in 0.000111 seconds (updated 0 objects)
```

jackspp, 2018-04-28:
Meet same problem. Hope someone can help solve it. Great thanks.

jd730, 2018-04-29:
@jackspp  Thanks for sharing your experience. I think I also checked using `nmap` and `netstat`. It is still closed although I set port 2000 or other.

nsubiron, 2018-04-30:
Does it work when you run the [compiled version](https://github.com/carla-simulator/carla/releases)?

And just to make sure, have you pressed "Play" after opening the editor? It doesn't seem so in the log.

jd730, 2018-04-30:
@nsubiron I am running on server. So I do not have play button. I succeeded one server using compiled version. However,  I still fail to use source code version. In addition the other server does not work both compiled and soource code

nsubiron, 2018-04-30:
@jd730 The source code version by default launches the editor, not the game, you can compile a packaged version on a headless server using

```
$ export UE4_ROOT=/Path/to/UnrealEngine
$ ./Setup.sh --no-editor
$ ./Package.sh
```

jd730, 2018-04-30:
@nsubiron  Thank you for giving information. 
Currently, I confronted CMake Error. I will resolve it later. 

By the way, one server had same problem on issues(port does not open) although I used compiled version.

cf. CMake error is below.
```
CMake Error: The current CMakeCache.txt directory /home/jdhwang/t_carla/Util/Build/googletest-source/build/CMakeCache.txt is different than the directory /ho                                                                                me/jdhwang/carla/Util/Build/googletest-source/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you ar                                                                                e not sure, reedit the CMakeCache.txt
CMake Error: The source "/home/jdhwang/t_carla/Util/Build/googletest-source/CMakeLists.txt" does not match the source "/home/jdhwang/carla/Util/Build/googlet                                                                                est-source/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
```

nsubiron, 2018-05-03:
@jd730 If the port does not open in just one machine, it's probably an issue with the configuration of the machine, make sure too that no other application in that machine is using the same port.

About the CMake error, I see I made a mistake in the instructions I gave you above, should be _Rebuild.sh_ instead of _Setup.sh_ (already fixed it), sorry about that :confused:. The easy but long way to fix it is to remove the folder _Util/Build_ and run the setup again, full instructions

```
$ rm -Rf Util/Build
$ ./Setup.sh
$ export UE4_ROOT=/Path/to/UnrealEngine
$ ./Rebuild.sh --no-editor
$ ./Package.sh
```

jd730, 2018-05-04:
@nsubiron  
Like you said, re-install worked although I used script of compiled version.
Thanks.

closed 2018-05-04 by jd730

SaiVinay007, 2019-06-01:
> @jd730
> I just checked the code for carla, find the error is raised in Pythonclient/carla/tcp.py, line 44:
> 
> `except socket.error as exception: 45 error = exception 46 logging.debug('%sconnection attempt %d: %s', self._logprefix, attempt, error) 47 time.sleep(1) `
> For me, this socket.error is caused by fact that carla server is not listening at port 2000,
> you could check whether server is listening or not by type
> 
> `lsof -i:2000` in terminal, if your port is 2000.
> 
> I use word-port 2000 to start the server, then the problem is solved.
> 
> Hope it's useful for you.

adding "-port=2000" helped
Thanks