carla-simulator/carla #379: Lighting artefact causing the road to change its brightness depending on the distance to the camera

YashBansod, 2018-04-26:
One of the bug fixes in the recent release (0.8.2) was:

- Fixed lighting artefact causing the road to change its brightness depending on the distance to the camera

The problem is fixed for the Town01 and Town02. But If I try to make my own map, the problem appears again.
Can you please guide me on how the problem was solved for the provided town maps?

TheNihilisticRobot, 2018-05-02:
Hi @YashBansod I haven't been able to replicate this issue in our pcs. This bug Fix was not Level dependant but asset dependant, so if your towns 1 and 2 are fixed all of them must be fixed.
Is there any chance that you are comparing Release versions of the maps and an editor version with non up-to-date assets?

YashBansod, 2018-05-02:
@TheNihilisticRobot  Thanks a lot  for the reply. I realized that the assets were not updated and I was still using the **Content_0.8.1**. I updated the assets to **Content_0.8.2** and the problem is resolved.

Don't know why but for some reason the problem wasn't happening for the default maps with the 0.8.1 assets in the editor too.

Updating the Contents resolved the problem.

closed 2018-05-02 by YashBansod