carla-simulator/carla #381: Pedestrians crossing the road get stuck

metaluga145, 2018-04-26:
Pedestrians sometimes get stuck in the middle of the road when a non-agent car comes close. This results in congestion on the road. Eventually all vehicles with autopilot turned on stuck in this congestion forever.
Described situation always repeats after 30-40 minutes of riding. 

Used parameters:
CARLA version: 0.8.1 / 0.8.2
Number of vehicles: 20
Number of pedestrians: 40
Map: default

ciumonk, 2018-04-30:
You can try adding code that checks whether the pedestrian has changed its position lately, either in AWalkerAIController::Tick(check all pedestrians every frame) or AWalkerSpawnerBase::Tick(check one pedestrian every frame, probably easier on the resources)..

juaxix, 2018-04-30:
This was a problem related to the issue #7 , as we are going to change the way the pedestrians walk and probably the navmesh because it is actually a bottleneck for the cpu (game) time, right now the walkers (pedestrians) use a basic behavior: if they see a car in the path they will stop and wait 5 seconds, after that, they will continue walking even if there is a car in front of them ,eventually they will crash against any vehicle and removed, but they will never wait forever in stuck state forever, so that might be a bug, can you send as a video of this particular case please?

This attached gif is the expected result of the case
[image]

Summarizing, the problem is more related to the way the vehicles work, and this is something more advanced, requires a bigger change in the cars AI.

metaluga145, 2018-05-04:
@juaxix please, do not close the issue for now. We are trying to collect data for video. However, our resources are busy right now, so it might take some time. I also run into another bug. Please see #399

closed 2018-08-06 by stale[bot]