carla-simulator/carla #383: Workflow for developing sensors - slow performance

mitchellspryn, 2018-04-29:
Hi all

I'm attempting to change the carla code base to collect data for one of my custom scenarios. I haven't changed any code yet (pulled master this afternoon). I've got the code building and running from within the Unreal Engine, but I'm seeing some odd behavior. When I run from within the unreal editor without the server (i.e. UseNetworking=false in CarlaSettings.ini), everything runs fine (fps > 60). When I run from within the editor with the server enabled (i.e. UseNetworking=true in CarlaSettings.ini), the simulator runs really slowly (<3 FPS). Also, I cannot get any sensor data - whenever I call client.read_data() (e.g. in client_example.py), I get an AssertionError (len(raw_data) < 4 * width * height). 

Is this expected? Ultimately, I'm just interested in exposing additional data streams in the python client, so I'd also be happy if there's another way to test that as well.

mitchellspryn, 2018-04-29:
So, I found a few interesting things:

1) The AssertionError is my fault. I was mixing the python client from the stable branch, and the code from the latest master. It's likely that the packet definition has changed. 

2) If I start the python client before starting the simulator in the editor, everything works as expected (I get ~20 FPS, which is fine). If I start the simulator, and then start the python client, everything is slow (<3 FPS). Does anyone know why this happens?

nsubiron, 2018-04-30:
Hi @mitchellspryn,

Is it possible that the low FPS happens when the editor doesn't have the focus?

From our FAQ

>  **Why do I have very low FPS when running the server in Unreal Editor?**
> UE4 Editor goes to a low performance mode when out of focus. It can be disabled in the editor preferences. Go to "Edit->Editor Preferences->Performance" and disable the "Use Less CPU When in Background" option.

mitchellspryn, 2018-05-01:
I did that already. 

Let me clarify my question. I'm not asking why the FPS is lower - I understand that running via the editor has an overhead. I'm asking why the perf hit is dependent on the order in which I start the server / client.

nsubiron, 2018-05-02:
I am not able to reproduce it, in our tests we get same performance no matter the order. I can't think of anything that could depend in the order in which you start the server and client.