carla-simulator/carla #384: Question : Why a vehicle has different speed in same condition?

jd730, 2018-04-29:
Hi,
I placed an agent on a starting point of one of the longest road with the same weather, no other vehicles, and pedestrians.  And I set turn right after N frames. However, these turning points are stark in contrast.

Could you tell me why it is different?

jd730, 2018-04-30:
I used two servers for testing speed.  I checked the number of frames from starting point to the end of a straight road. One is 200 frames including init. The other is 500 frames. Do forward_speed and other measurements which are related to time consider server time or performance??

jd730, 2018-04-30:
I guess the answer of speed issue is on #384 (fixed frme fate) I will try right now.

jd730, 2018-04-30:
@nsubiron  How about starting orientation issue?
[NissanMica](https://youtu.be/mDUUeI__Q0Y)
[Mustang](https://youtu.be/6grCkr3hq_o)

White column is Nissan and gray column is Mustang
[image]

nsubiron, 2018-05-03:
@jd730 The dynamics of the vehicles are simulated by the physics engine, and they are different for each vehicle (different physics asset based on the mesh, different engine setup, wheel distance, weight, etc), so the resulting position is going to vary a bit.

Nevertheless, it seems the Micra has some issue, it shouldn't deviate that much from the straight-line. I'm going to open an issue for this.