carla-simulator/carla #385: Windows binares of 0.8.2 version

eds89, 2018-04-29:
Hello,
Please can you provide binaries for version 0.8.2? The PythonClient that comes with version 0.8.1 no longer works. Some (but not all) of the changes commited to version 0.8.2 were already included in version 0.8.1. 

Unfortunately I do not have enought resources nor skills to build it myself.

Regards,

juaxix, 2018-05-02:
Hello @eds89 , the [Windows version for CARLA 0.8.2](https://github.com/carla-simulator/carla/releases/tag/0.8.2) is now released. Please report any issue you may encounter as this version is still experimental.
Thanks

eds89, 2018-05-02:
Hello,
Many thanks!

Have you used the same Windows build files provided in Issue #21 to build
this version?

juaxix, 2018-05-02:
We use the files from the master branch with the tag associated to its version. For example, for that particular build it uses the files from the [tag 0.82](https://github.com/carla-simulator/carla/tree/0.8.2/)

eds89, 2018-05-02:
In other words: what build scripts have you used to compile version 0.8.2?

juaxix, 2018-05-03:
To build 0.8.2 using Visual Studio 2017 you have to use 

> make vsproject15

 in the main folder, then open CARLA solution and build it for release, then open the Unreal project and build it, and use Unreal to Package.

juaxix, 2018-05-03:
@wzhAptiv hi. It may take a while but you may want to look at the logs, it should end eventually...

wzhAptiv, 2018-05-03:
@juaxix I've been waiting it for more than 3 hours... Is this the case? or something went wrong but not logged? The log only shows it is stuck at [16/17], nothing further happened until I close the window, and then compiling fails. 
It's quite wired...

juaxix, 2018-05-03:
If the compilation stops it's probably because it failed somewhere inside one of the macros Unreal has to compile, if this is the case, there is no error, you have to re-make CARLA plugin project, rebuild it, and regenerate the unreal project + rebuild everything

wzhAptiv, 2018-05-03:
@juaxix Thank you for the information and help. I'll try to rebuild everything, or switch to another compiler/platform.

eds89, 2018-05-04:
Hi. Did you compile it on Windows?

juaxix, 2018-05-04:
Yes @eds89, with vs 2017

eds89, 2018-05-05:
Hi @juaxix, I get no message when running 'make' on the CARLA folder. The command is on the %PATH% variable, but there is no output.

juaxix, 2018-05-07:
You have to [make](http://gnuwin32.sourceforge.net/packages/make.htm) sure that you have make installed. 
The log should be something like this

`E:\CARLA>make vsproject15
Compiling carla_server.proto...
done.
-- Selecting Windows SDK version 10.0.16299.0 to target Windows 10.0.15063.
-- Configuring done
-- Generating done
-- Build files have been written to: E:/CARLA/Util/Build/carlaserver-build/visualstudio`

closed 2018-06-04 by juaxix

eds89, 2018-09-13:
Hello,
When I run:
``
make vsproject15
``
I get
``
F:\CARLA082_Dev>make vsproject15
make[2]: Entering directory `F:/CARLA082_Dev'
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

F:\CARLA082_Dev>
``
Do you have any idea what I may be doing wrong? Cmake is on the PATH for sure.

eds89, 2019-01-25:
Hello. Where's nmake called within the Makefile? I can only find references to Cmake, not nmake, even though I'm getting the following error:
`NMAKE : fatal error U1065: invalid option 'H'`