carla-simulator/carla #387: .Net client (as well as starting point for other programming languages)

Soolek, 2018-04-30:
Hi,

First of all THANK YOU for making Carla such a great autonomous car developing platform. With recent Lidar implementation and fps optimizations it's a perfect starting point for new autonomous car developers and/or enthusiasts.

I would like to implement a native .Net client for Carla since this is the framework I feel most comfortable to work with [(ADAS example)](https://github.com/soolek/autocruise). I have tried to run the PythonClient using IronPython as a quick workaround but got stuck on including dependencies, so I have decided not to rush it and implement it properly.

To make it adhere to best coding standards I would like to make it as easy to support in future as possible - for example by self generating classes based on some kind of protocol config file?

TL;DR **What would be the best starting point for implementing Carla client in other languages/frameworks?**
1) Documentation on existing server-client protocol
2) What should be the basis to make it flexible enough for future protocol changes?

Any suggestions are welcome.

nsubiron, 2018-04-30:
Hi @Soolek,

Yes, it is a great idea, and it would be very nice if you can contribute to this, **but, we are about to do major changes in the server-client protocol.** So perhaps it's better if you wait a bit until we have a clearer plan for the new interface. The current design is going to give a lot of scalability issues in the long run, so we want to change some aspects now to support a more flexible API, i.e., multi-client, more control over all the agents, sensors, etc. I'm going to be in charge of these changes, I will give more details in the coming weeks. Now we are defining a new protocol, once we have a clear definition you can start developing an .NET interface for it. I will keep you posted ;)

Soolek, 2018-04-30:
Sounds good - Thank you. Please do keep me posted.

I'm about to submit an abstract about autonomous driving development platforms to ICSES18 that will take place on the 10th of September. I would like to show Carla along with a working example by then, hence the need for a .Net client.

With your current development speed I do believe time won't be an issue :)

closed 2018-05-02 by Soolek