carla-simulator/carla #388: Multi-lane roads

Soolek, 2018-04-30:
Hi,

I have noticed that the roads in Carla are based on single, counter traffic lanes. I would like to expand existing maps (or add new one for fps sake) which might be useful for all kinds of highway scenarios ([example](https://youtu.be/lVlqggTiTzY?t=336)).

To proceed:
1) I have read that there is some kind of traffic flow generator, but does it support multiple same direction lanes?
2) Does the generator support multiple level roads? (for highway entry/exit bridge merge scenarios)
3) Does traffic support random lane changes?
4) Are any specific requirements needed to be met before importing into Carla?

Since I don't posses needed 3d skills for creating such map, I would like to outsource this task - therefore specific requirements are needed for an order.

felipecode, 2018-05-02:
Hello Soolek !
That seems very cool that you are whiling to contribute to CARLA
Regarding your questions. These are things we are working
right now. But before working on them we are working on adopting standards for road
and scenario definitions. 

We will be posting more details about the new CARLA architecture in
a near future.

Soolek, 2018-05-02:
Sounds great!
I will make appropriate arrangements as soon as multi-lane roads requirements are known.

closed 2018-05-02 by Soolek