carla-simulator/carla #390: client.start_episode() takes so~~~ long (4.5 s) Does it have any solution?

jd730, 2018-05-03:
Hi,
Does it have any solution to make client.start_episode() faster?
Currently, it takes 4.3~4.6 seconds so that I could not learn my agent using many iterations.
If I need to 10k episodes I need to wait 11 hours.

Thank you.

nsubiron, 2018-05-03:
Hi @jd730,

When you request a new episode the simulator has to reload the scene and spawn all the agents, this is a slow operation, I don't think there is anything easy to do to speed up this process. Saving the scene would probably be as slow, the scene is very complex and has many objects to load.

jd730, 2018-05-03:
It seems that CARLA server does not use multi-core cpu but single core, right??
When I tried to I7-7700K, it takes 2.5~3 s.
c.f.) above result is conducted on Zeon

jd730, 2018-05-04:
@nsubiron  Is speed of benchmark server different from that of general CARLA server?

nsubiron, 2018-05-08:
The simulator uses multi-core, but I'm afraid most of the scene loading code is executed on a single thread. This is all handled by Unreal Engine.

I'm opening an issue for tracking purposes, not a high priority right now but we would like to optimize the scene loading time eventually.

>  Is speed of benchmark server different from that of general CARLA server?

Do you mean running the simulator with the `-benchmark` flag? If so, the net render speed is the same, but if you set a very low FPS then you get less images per second thus you can simulate more seconds in less time. See [fixed frame-rate](http://carla.readthedocs.io/en/latest/configuring_the_simulation/).

jd730, 2018-05-08:
@nsubiron  Thanks a lot.
What I mean is the communication speed between server and client. I wondered whether there is some optimized. When I took a look at the code. I could not find it.

 From the reason, as you mentioned above, we set fps as 8. However, it is still slow so that we use 5 servers for one agent using multithreading libraries. So I was curious how to train the model. I guess just make several workers.(A3C)

nsubiron, 2018-05-08:
If you mean optimizing the speed in which the data is sent through the network, there are [some tweaks](https://wwwx.cs.unc.edu/~sparkst/howto/network_tuning.php) at system level that may improve that. But usually the main bottleneck of CARLA is the rendering of the images.

closed 2018-05-22 by nsubiron