carla-simulator/carla #391: Time out problem when running autopilot demo

Liangsyl, 2018-05-03:
Hi, i am using CarlaUE4 which is built on Windows 10 envirimental.
And now, i am tring to follow the direction "Connecting a Python client".
I am using VS2015 x64 native tools command prompt to run the client example, and I can start running the autopilot example, but after it run a while, time out error occured, and the prompt showed that the distance host forced closing a existing connection.
After a while, a new episode started itself, and it did not run too long, the time out error occured agian.
[image]

[image]
 
It never autopilot successfully.
Do you have any suggestions or way to solve this problem?

Liangsyl, 2018-05-07:
hi，@marcgpuig.I am using CarlaUE4 0.8 which is built on Windows 10 environment.
I am tring ro run driving_benchmark_example.py. I opened _$./CarlaUE4.exe -carla-server_ at first, then, i opened _$./driving_benchmark_example.py_.
However, time out problem occured all the time, and it never connected sussesfully. 
And I tried to run _$./client_example.py --autopilot_, it can connected sussesfully.
What can i do to solve this problem? 
[image]

felipecode, 2018-05-10:
@marcgpuig  @juaxix 
Can someone reproduce this in a windows system ?

juaxix, 2018-05-10:
Hello all. @Liangsyl  i think there is a problem with the connection, looking to the errors: 10054,  10061...have you allowed the .exe files in your firewall?, have you tried using a different port instead of 2000? ([CarlaSettings.ini](https://github.com/carla-simulator/carla/blob/master/Docs/Example.CarlaSettings.ini#L16)) ,maybe you have this port used by other software.

closed 2018-08-06 by stale[bot]