carla-simulator/carla #393: Where does CARLA store the videos recorded?

ppalaupuigdevall, 2018-05-03:
Hello,

I am using CARLA simulator on Ubuntu 16.04. I want to record a video of "MyCamera", which is a camera that I enabled in the file _Example.CarlaSetiings.ini_ . This file looks like:
`; Example of settings file for CARLA.
;
; This file can be loaded with the Python client to be sent to the server. It
; defines the parameters to be used when requesting a new episode.
;
; Note that server specific variables are only loaded when launching the
; simulator. Use it with `./CarlaUE4.sh -carla-settings=Path/To/This/File`.

[CARLA/Server]
; If set to false, a mock controller will be used instead of waiting for a real
; client to connect. (Server only)
UseNetworking=false
; Ports to use for the server-client communication. This can be overridden by
; the command-line switch `-world-port=N`, write and read ports will be set to
; N+1 and N+2 respectively. (Server only)
WorldPort=2000
; Time-out in milliseconds for the networking operations. (Server only)
ServerTimeOut=10000
; In synchronous mode, CARLA waits every frame until the control from the client
; is received.
SynchronousMode=true
; Send info about every non-player agent in the scene every frame, the
; information is attached to the measurements message. This includes other
; vehicles, pedestrians and traffic signs. Disabled by default to improve
; performance.
SendNonPlayerAgentsInfo=false

[CARLA/QualitySettings]
; Quality level of the graphics, a lower level makes the simulation run
; considerably faster. Available: Low or Epic.
QualityLevel=Epic

[CARLA/LevelSettings]
; Path of the vehicle class to be used for the player. Leave empty for default.
; Paths follow the pattern "/Game/Blueprints/Vehicles/Mustang/Mustang.Mustang_C"
PlayerVehicle=
; Number of non-player vehicles to be spawned into the level.
NumberOfVehicles=0
; Number of non-player pedestrians to be spawned into the level.
NumberOfPedestrians=0
; Index of the weather/lighting presets to use. If negative, the default presets
; of the map will be used.
WeatherId=10
; Seeds for the pseudo-random number generators.
SeedVehicles=123456789
SeedPedestrians=123456789

[CARLA/Sensor]
; Names of the sensors to be attached to the player, comma-separated, each of
; them should be defined in its own subsection.

; Uncomment next line to add a camera called MyCamera to the vehicle
Sensors=MyCamera

; or uncomment next line to add a camera and a Lidar
; Sensors=MyCamera,MyLidar

; or uncomment next line to add a regular camera and a depth camera
;Sensors=MyCamera,MyCamera/Depth

; Now, every camera we added needs to be defined it in its own subsection.
[CARLA/Sensor/MyCamera]
; Type of the sensor. The available types are:
;   * CAMERA                        A scene capture camera.
;   * LIDAR_RAY_CAST                A Lidar implementation based on ray-casting.
**SensorType=CAMERA**
; Post-processing effect to be applied to this camera. Valid values:
;   * None                  No effects applied.
;   * SceneFinal            Post-processing present at scene (bloom, fog, etc).
;   * Depth                 Depth map ground-truth only.
;   * SemanticSegmentation  Semantic segmentation ground-truth only.
PostProcessing=SceneFinal
; Size of the captured image in pixels.
ImageSizeX=1344
ImageSizeY=968
; Camera (horizontal) field of view in degrees.
FOV=145
; Position of the camera relative to the car in meters.
PositionX=0.20
PositionY=0
PositionZ=1.30
; Rotation of the camera relative to the car in degrees.
RotationPitch=8
RotationRoll=0
RotationYaw=0

[CARLA/Sensor/MyCamera/Depth]
; The sensor can be defined in a subsection of MyCamera so it inherits the
; values in MyCamera. This adds a camera similar to MyCamera but generating
; depth map images instead.
PostProcessing=Depth

[CARLA/Sensor/MyLidar]
SensorType=LIDAR_RAY_CAST
; Number of lasers.
Channels=32
; Measure distance in meters.
Range=50.0
; Points generated by all lasers per second.
PointsPerSecond=100000
; Lidar rotation frequency.
RotationFrequency=10
; Upper and lower laser angles, positive values means above horizontal line.
UpperFOVLimit=10
LowerFOVLimit=-30
; Position and rotation relative to the vehicle.
PositionX=0
PositionY=0
PositionZ=1.40
RotationPitch=0
RotationYaw=0
RotationRoll=0
`
The text part in bold is the uncommented line for adding a camera called MyCamera. My question is:

**How can I record video from this camera and where is this video recorded?**

Thanks in advance,

jd730, 2018-05-03:
You can easily implement to record video using skvideo.io

closed 2018-05-24 by marcgpuig

anishgorantiwar, 2019-07-23:
Hello, I had the same question, and I wasn't able to capture the video using skvideo.io. Can anyone help me with that.
Thanks