carla-simulator/carla #394: Is there any guide to intergrate logitech g29 to carla 0.8.1 or 0.8.2? Please guide me to the available resource.

zeroenergyhere, 2018-05-03:


wzhAptiv, 2018-05-03:
pygame should support G29 after you install its driver. 
use `get_axis` to get current status of wheel and pedals. 

Then, try to modify or extend `_get_keyboard_control` function in `manual_control.py`

more API can be found from [pygame document](https://www.pygame.org/docs/ref/joystick.html)

zeroenergyhere, 2018-05-04:
Thank you for the reply, i will try the same.

zeroenergyhere, 2018-05-04:
I have installed pygame and cloned your pygame-manual_control library to PythonClient. But I am unable to control the car using steering wheel or manually.

wzhAptiv, 2018-05-04:
Hi @zeroenergyhere , you need to first run CARLA in server mode, and be sure your firewall is not blocking the port, default is 2000. Then you run the python script, but be sure having window focus on the popped window. 
Firstly, you should try the python script `PythonClient/manual_control.py` provided in the repo, and be sure you can use keyboard to control. Then, you can modify or extend the script for steering wheel.

LijieLiu, 2018-05-15:
Hi @wzhAptiv 
Im trying on using logitech G920 to control the vehicle, but seems like there are some issue at manual-control.py. I ran manual-control script as client but it showed 
[image]
And the CARLA is running on Unreal Editor
Any idea?
Thanks

wzhAptiv, 2018-05-15:
@LijieLiu remote desktop (VNC) normally doesn't support graphic card rendering, either try direct using the machine, or find a remote desktop software supporting graphic card (supporting X11 for example). I don't have much experience in remote desktops, so I don't have any recommendations.

felipecode, 2018-05-15:
Hey !
CARLA does not yet support using joysticks. 
But we are currently working with people in a pull request for that.
Please follow issue #88 .

LijieLiu, 2018-05-15:
Hi! @wzhAptiv 
Thanks for reply. I was misled to use bash instead of cmd on it, my bad. Now I can run manual-control with cmd without error. However, WASD keys can not control the vehicle, but arrow up down left right can change the direction of camera, and enter key can make vehicle jump. Is there any problem with my pygame or what? I am not sure. Thanks a lot
Hi @felipecode 
Thanks for letting me know that. will have a look.

wzhAptiv, 2018-05-15:
@LijieLiu Have your window focus on the pygame window, instead of the server window. The pygame cannot receive keyboard input without window focus.

wzhAptiv, 2018-05-15:
@LijieLiu Hi, I haven't tried customized map. But the compiled binary + original manual_control.py works fine for me, both on Windows and Ubuntu. Maybe you can look into the function `_get_keyboard_control()` either debugging or printing to check if the client is correctly sending the control parameters to the server.

LijieLiu, 2018-05-20:
@wzhAptiv 
OK thanks anyway. I will have a try on this function.
THX

closed 2018-08-06 by stale[bot]

0102TH, 2019-03-31:
Hello, how is the process? I am currently doing the integration of the logitechG920 into the carla server. But there was a problem running manual_control_steeringwheel.py. So I also searched for answers according to the displayed questions. One thing is that after I installed jstest-gtk, I opened it, but nothing was displayed, which made me very confused. Can someone give me some advice, thank you very much.

The following is the message that prompted the error.
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server 192.168.1.7:2000

Welcome to CARLA manual control with steering wheel Logitech G29.
To drive start by preshing the brake pedal.
Change your wheel_config.ini according to your steering wheel.
To find out the values of your steering wheel use jstest-gtk in Ubuntu.

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.4-84-g889b998-dirty 
WARNING: Simulator API version  = 0.9.4-118-g3c18dee-dirty 
Traceback (most recent call last):
  File "./manual_control_steeringwheel.py", line 859, in <module>
    main()
  File "./manual_control_steeringwheel.py", line 851, in main
    game_loop(args)
  File "./manual_control_steeringwheel.py", line 781, in game_loop
    controller = DualControl(world, args.autopilot)
  File "./manual_control_steeringwheel.py", line 225, in __init__
    self._joystick = pygame.joystick.Joystick(0)
pygame.error: Invalid joystick device number

Derekabc, 2019-05-17:
hi @0102TH , do you know how to set up to use the G920 by manual_control_steeringwheel.py. Thanks

soulslicer, 2019-10-16:
Is there anything that works with the recent version of CARLA?