carla-simulator/carla #399: Car in the bush!

metaluga145, 2018-05-04:
Hi there! During experiments we run into another bug. It is probably related to car navigations system.
Please, see the image :)
We do not know how that happened. All cars in the simulator were on autopilot. Agent car arrived already after the red car had stuck in the bush.
Carla version: 0.8.2

Thank you! :)
[image]

marcgpuig, 2018-05-08:
Hi @metaluga145 

Are you running the simulation on fixed framerate?
If it is the case and you are experimenting low FPS, it is possible that your physics engine have no time to update properly.
This may cause this kind of glitches.

metaluga145, 2018-05-08:
Hi @marcgpuig 

No, the simulator was not running on fixed framerate.
The engine was running in the server mode without -benchmark flag.
As for the client side I used script quite similar to point_cloud_example.py.

As you can see from the image, there are few cars ahead. The whole road in front and behind my agent stuck in congestion. I waited for minutes hoping that the red car will somehow get out of the bush, but then I just stopped the simulation.

nsubiron, 2018-05-08:
Ok, there are two modes to run the simulator, and two modes to keep the client-server communication.

  * Fixed frame-rate vs variable frame-rate (chose when launching the simulator, `-benchmark` flag).
  * Synchronous mode vs asynchronous mode (chose at the beginning of an episode in the CarlaSettings).

How fixed frame-rate works is explained [here](http://carla.readthedocs.io/en/latest/configuring_the_simulation/).

But the details of the sync vs async are missing, I'll try to get time this week to write that. In sum, in  synchronous mode the simulator waits for the client to send the control message every frame. **If you use synchronous mode the simulator should always run at fixed time-step**, i.e., with `-benchmark` flag. Otherwise, if the client is too slow, like it is when saving the point cloud, the simulator freezes and next time the physics engine has to update it will have to simulate a very long step. During this long step there will be no autopilot update so the vehicles can end up not turning in time (plus the physics might do weird things if the step is too long).

Long story short, you should run the simulator with the `-benchmark` flag ;)

metaluga145, 2018-05-08:
@nsubiron Thank you for the answer. That was quite helpful :) 
We would highly appreciate an update for synchronous and asynchronous mode documentation :) 

This issue can be considered as a question and you can close it :)

closed 2018-05-08 by nsubiron