carla-simulator/carla #404: Repeatability over multiple episodes

mtikekar, 2018-05-05:
I am using Carla 0.8.2 on Linux. The following steps will work to reproduce the bug:

1. Run the server and client for 3 episodes like in `client_example.py` and save the settings for each episode as `settings0.ini`, `settings1.ini` and `settings2.ini`
2. Restart server and client and load `settings0.ini`
3. Restart server and client and load `settings1.ini`
4. Restart server and client and load `settings2.ini`

I find that I get repeatable data only for the first episode. For the other episodes:
1. List of starting positions returned by `client.load_settings` is shuffled
2. Pedestrians and vehicles are initialized differently

I can fix the first problem by going through the entire list and selecting the same starting position as before. But it would be much easier if the list was not reordered every time, so that I only need to save the index.

For the second problem, it seems there is something more than `SeedVehicles` and `SeedPedestrians` that is controlling the initialization.

felipecode, 2018-05-05:
Hey @mtikekar 

This bug seems very related to #310,
Before the first reset , the starting positions and objects positions are different. After you do the first reset,  the starting points and objects positions are likely to be stable.

Can you confirm that ? 

Thanks !

mtikekar, 2018-05-05:
Yes, it seems that I can get perfect repeatability if I do something like the following for each episode:

```python
# reset once
client.load_settings(settings)
client.start_episode(0)

# actual start
scene = client.load_settings(settings)
client.start_episode(...)
```

I hope this can be fixed so that the reset is not needed. But, until then, I'll do the reset.

closed 2018-08-06 by stale[bot]