carla-simulator/carla #405: the autopilot drive mode in carla-client_side_agent

Fengmoon93, 2018-05-07:
1.I solved all the error in the autopilot.py and I can run it
but I found the control is just to speed up and hit the wall
I saw in the autopilot.py

# TODO: Do not use the pilot configuration python file
from carla.autopilot.pilotconfiguration import ConfigAutopilot

and the note:maybe  the autopilot won't work in the Python3
em....I don't know how to solve it.

I used to collect the driving data by using the 
control = measurements.player_measurements.autopilot_control
but this time the task is to set the start and goal
2.So,I collect the data by Pygame control.and I found sometimes the forward speed <0?

jackspp, 2018-05-08:
@Fengmoon93 

When I use autopilot drive mode in carla-client_side_agent, it starts from top left sky and keep starting new episodes...  Do you know how to solve this problem?

Fengmoon93, 2018-05-08:
I think it's the pose, you can select 2 points as the start and the goal by yourself to see the problem is still there.

Fengmoon93, 2018-05-08:
I solve the problem,just use the 0.7 
it works......

Fengmoon93, 2018-05-08:
thanks for your reply
waiting for that.....

closed 2018-05-08 by Fengmoon93