carla-simulator/carla #410: Autopilot question

eds89, 2018-05-10:
Hello,

Running autopilot has been a disappointing experience. With it turned on, the only thing I get is the car driving in a straight line until crashing on the wall. Occasionaly, it does attempt to make curves, but it doesn't get beyond that.

YashBansod, 2018-05-11:
Hello @eds89, can you tell if you are running CARLA in synchronous or asynchronous mode? Also are you running CARLA at fixed frame rate? if so, what is the frame rate you are running at?

nsubiron, 2018-05-11:
For the autopilot to work, the commands need to reach the server on time; so either the client is fast enough to follow the frame-rate of the simulator or you use synchronous mode.

Also see this https://github.com/carla-simulator/carla/issues/399#issuecomment-387362225.

closed 2018-05-22 by nsubiron