carla-simulator/carla #412: Multiplayer Ability

akhil22, 2018-05-11:
Hi,
I was wondering if we can have multiple instances of the environment running on different machines with each machine simulating n number for vehicles. The environment will be shared between all the machines over a network just like in a multiplayer game.

nsubiron, 2018-05-14:
Hi @akhil22,

What we are working now is to extend the networking library so multiple Python clients can connect to a single simulator, the idea is that each client receives only what it needs instead of every client receiving everything. But this would go outside Unreal environment. The "control" communication is done with TCP to have the option to block until a client sends the control. The synchronization issues can be resolved by running the simulator with [fixed time-step](http://carla.readthedocs.io/en/latest/configuring_the_simulation/), this way the simulation is stable even if it has to wait a long time for a client to reply. We are also studying using UDP for the sensor data (images) to gain some speed, although so far the network load doesn't seem to be a bottleneck. Using shared memory is not in our short term plans, and in the case that we decide to do it, it would be as optional as we want the ability to use different machines.

If I understood correctly, what you want is to have multiple simulators rendering images, like a multiplayer videogame. This can be achieved with Unreal Engine's built-in replication system. We have also studied this possibility, but it does not seem to be trivial to implement; you need to fine tune your replication for all the Unreal actors playing a role in the scene. The network replication is done through UDP so synchronization can also be tricky if a client is too slow. For now we decided not to implement this, however it might be something we eventually need (it would be great to have this for scalability).

akhil22, 2018-05-14:
Hi Nsubiron,
Thanks for you reply. The main reason we want to have multiple clients sharing the scene which are rendering independently is the bottleneck in sending images and lidar data over UDP. If each client can render the scene they can simulate lidar and camera, we don't have to send this huge data over the network. if you have 10-20 clients then simulating all of their sensor on one machine and sending this data over network (including copying the image data from GPU) will significantly effect the frame rate . Unity has a parameter which allows sending the transformation of the objects to a particular client only if that object is in the frustum of any of the cameras rendering in that client, this significantly reduces the data sent over the network. I believe Unreal would also have something like this.  For synchronization, either we can use ntpdate which can synchronize the systems clocks up to 1ms(+-) error(Currently using this with Unity) or we can have a simulation clocks running on one server and sending the time stamp to all the clients (just like ROS with use_sim_time as true). I also have one more question, since Unreal is written in c++ can't directly create nodes that publish sensor data without using websockets (I understand you want this data to be sent to other clients, but for the same machine this seems to be more efficient to me)?

nsubiron, 2018-05-30:
Yes, this multiplayer setup has definitively great advantages and the gain in performance can be very big. It's something we would like to have in Carla eventually, so any progress you make on this please report :)

And yes, as you say, in the C++ classes you can do anything you want, like using shared memory. But in that case I doubt you can use the built-in replication framework, at least not without modifying the engine's source.

The only thing stopping us from implementing these things now is that they involve significant complexity and we are a very small team to code and maintain these changes. There are many things we want to fix/improve before getting into something this big. In any case, I'm adding this to our backlog.

germanros1987, 2020-03-19:
This is related to the future multi-server architecture. We will follow up when we are ready to start.

closed 2020-06-26 by germanros1987

gouchaoer, 2022-01-24:
@germanros1987 any progress? thanks for carla team!