carla-simulator/carla #413: feature request for adding snow condition

hc167, 2018-05-11:
After slack chat discussion at udacity lyft-challenge channel. Look like it is a good feature if CARLA can support snowy day condition.

germanros1987, 2018-05-21:
I think we should be adding general road conditions such as: rain, snow, ice, etc. I am adding it to our roadmap to do more realistic control simulation. Thanks!

berkerlogoglu, 2019-11-13:
how about effect of rain, snow and fog on lidar returns?

germanros1987, 2020-03-19:
Added to 0.9.12 sprint.

XGodina, 2020-07-07:
@berkerlogoglu, @hc167 Hi, we keep in mind to do that. But now we are working to set the new Sky atmosphere in CARLA. As German said we will add to 0.9.12. sprint.

otoofim, 2021-08-12:
Hi, as 0.9.12 released, just wanted to ask if the weather features (rain, snow, ice, etc) are added? I am asking because I couldn't find any information about weather features update on the release page. Thanks.

daniel-bogdoll, 2022-05-11:
I also wanted to ping this issue, since affected lidar/radar measurements would be really useful

jiankebe, 2023-03-10:
> I also wanted to ping this issue, since affected lidar/radar measurements would be really useful

as 0.9.13 released, Has it been resolved？

leishi23, 2024-03-03:
Still no snowy weather yet - CARLA 0.9.15