carla-simulator/carla #414: Low fps for Carla 0.8.2 Server Mode

LijieLiu, 2018-05-12:
Hi!
I've successfully built Carla0.8.2 on windows, and it can work by standalone mode with Unreal Engine Editor now, and the fps is approximately 120. But when I tried it on server mode, the fps is 3 and the screen is stuck. 
[image]

I have disabled  thee "Use Less CPU When in Background" option for Unreal Engine Editor but nothing changed.
[image]

 I have tried on build exe file for CarlUE4, because seems like it can launch exe file with desirable size of window when launch exe file. But Unreal Engine Editor crushed while building. Any idea to solve that? Thanks a lot!

juaxix, 2018-05-14:
Hello @LijieLiu ,don't use the server only mode "Run Dedicated Server" , because it's not ready for networking yet. 
I'm curious about the crash when you build the project. 
If you build the game and you run it , it's no longer the editor.
Can you send me the full log of the cooking process and the log from the crash when you run it?

LijieLiu, 2018-05-14:
Hi @juaxix 
Thanks for reply. Sorry I am not very clear about the means of 'run dedicated server". I am working on using Logitech G920 Driving Force to control the vehicle, and seems like it can only work on server mode, so I have modified the manual-control.py to make it work on Logitech Driving Force. I created a CarlaSettings.ini file and set true to UseNetwork, as F.A.Q said [http://carla.readthedocs.io/en/latest/faq/](url), then I click play on Unreal Editor and run manual-control, but it keeps showing that port 2000 error. 
Is there any issue on this? I am not sure, please let me know.
And I will update the crash log later. Thanks a lot.

juaxix, 2018-05-14:
Ok @LijieLiu ,that's the simulator server, not the Unreal server, I mean, if you use the flag 
_UseNetwork_ ,and run the game (Unreal Engine project) internally, you are creating a socket by the CARLA plugin, so it's opening a port, and you have a Simulator Server up ,running and listening in that port: 2000. 
Now, if you have a problem with the port, what is this error displayed?, go and change this number in the settings, because it's probable that the port is being used by other program or you have the UE or the programs blocked in the firewall, check this out first.

LijieLiu, 2018-05-15:
Hi @juaxix 
Seems like it is firewall issue, and problem solved. Thanks a million!

closed 2018-05-15 by juaxix