carla-simulator/carla #418: Download fail

raulortiz8, 2018-05-15:
Hello, I am trying to download the compiled version for linux. But there is an error downloading it. In the download name you can read carla.tar.gz.part, this means that only a part has been downloaded, and I can not extract it.
Thanks in advance

juaxix, 2018-05-15:
Hello @raulortiz8 , that's probably because the download connection failed at some point that's why you have a .part file; you can try to use a plugin for your browser that handles downloads in a better way so when it fails ,it will try to continue and recover the file until the download is done.

raulortiz8, 2018-05-15:
Hello, thanks for your reply. But I can download the source code, for
instance. And whatever I want to download , but I do not know what happens
with this file. Could someone send me by dropbox or whatever.
Thanks in advance.

juaxix, 2018-05-16:
@raulortiz8  Can you try to use another connection via VPN?, maybe your country is blocking google drive ,where it is currently stored the releases...what is the connection error btw?

closed 2018-05-30 by juaxix