carla-simulator/carla #419: How do i change the npc's model

lockeregg, 2018-05-16:
Hi,
I was wondering if we can change the npc's model.

lockeregg, 2018-05-22:
Hello,
Any ideas?

TheNihilisticRobot, 2018-05-22:
Hi @lockeregg, sorry for the late response.

Sure, you may modify any NPC model you want. In [our documentation](http://carla.readthedocs.io/en/latest/how_to_add_assets/) you can find instructions on how to create and add your own vehicles (or any vehicle you have) to the project. 

If you Want to change the pedestrian models you'll have to create a child of the **Walker** class (Game/Blueprints/Walker) there, select the component **Mesh** And replace the Skeletal mesh and the Animation with your own.

[image]

After you've done that, you'll need to change the **WalkerSpawner** in the same folder to make it spawn your new character while playing. Open it and Modify the variable **AutomatedClasses** adding all the characters you need.

[image]

lockeregg, 2018-05-29:
Thank you very much!
What about motorcycle?
How can I change it to the motorcycle model?

Thank you!

TheNihilisticRobot, 2018-05-30:
So, we'll have our own bike and motorbike models to use with Carla in the next release which is already near. You'll be able to use our bike models or add your own easily!

lockeregg, 2018-05-31:
Great!!
Thank you very much!

closed 2018-05-31 by marcgpuig